FreeCAD/FreeCAD #14148: Automatic python modules documentation is broken

krushia, 2024-05-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

`Help > Automatic python modules documentation` Opens browser to sad page.

> FreeCAD Documentation	 
HTMLDoc.heading() takes from 2 to 3 positional arguments but 4 were given

[image]

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36
  * A2plus 0.4.64a
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * fasteners 0.5.20
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools
  * workfeature
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.5.3
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

None

### Anything else?

Breaks in same way on native (non-AppImage) 0.21.2 Install.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-19:
It opens the system browser on my side, everything working as expected with yesterdays weekly. Please report back if this issue still persists.

krushia, 2024-05-19:
I am using yesterday's weekly.

maxwxyz, 2024-05-19:
could it be related to general issues with the Appimage?

krushia, 2024-05-19:
https://github.com/FreeCAD/FreeCAD/blob/9decb98c71fe7b3299d422c2b0b92f0507c11c91/src/Gui/OnlineDocumentation.cpp#L130-L132

Assuming this is the problem, I'm not seeing the 3 arguments here match up with the pydoc function:

https://github.com/python/cpython/blob/30c687cdf8f7073302e92b2b125553d518a94eac/Lib/pydoc.py#L606

@maxwxyz Are you sure this works on your end? Are you on Windows?

maxwxyz, 2024-05-19:
I'm on Windows, yes.

krushia, 2024-05-19:
It seems that specifically the '/' page is broken. http://localhost:7465/

@maxwxyz Can you check that specific URL after summoning `Help > Automatic python modules documentation`? If that specific page works for you, then there might be a deeper bug between QByteArray and Python. It really shouldn't work with the code I'm seeing.

maxwxyz, 2024-05-19:
I've wiped my install and downloaded the latest weekly again and now I can reproduce. http://localhost:7465/

krushia, 2024-05-19:
This is a super easy fix, so I'd like to take a shot at it. Trying to get a working dev environment setup now.

maxwxyz, 2024-05-20:
Let me know if you have issues, otherwise I can create a PR before the freeze.

wwmayer, 2024-05-23:
Works fine with 
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37462 (Git)
Build type: debug
Branch: main
Hash: d12635246f86947031776a42714028b236bd05ed
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)

Could be related to the Python version. What's the output if you type this into the Python console:
```
import pydoc
help(pydoc.html.heading)
```

maxwxyz, 2024-05-23:
@wwmayer Not OP but this is the result when I click the command:
[image]

This is the result:
[image]

wwmayer, 2024-05-23:
I have checked the heading() function and two arguments have been dropped. I am working on this right now...