FreeCAD/FreeCAD #14150: [Feature Request] Sketcher: auto-switch distance constraint to driven if it overconstrains.

FlachyJoe, 2024-05-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Instead of create an over-constrained sketch when adding a supernumerary constraint it could be auto-switched to driven.
Moreover, when **Auto remove redundant** is checked the just created constraint is currently totally removed.

### Full version info

```shell
all
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-05-22:
This is a great idea!
It's always good if user actions DO NOT lead to data loss (after all, this is work). Especially if it is an option.

pierreporte, 2024-08-31:
I don’t agree with this. It is sometimes desired to keep the sketch overconstrained when changing how the sketch is constrained. The redundancy would disappear itself when the other dimensions or constrains are deleted.

FlachyJoe, 2024-09-01:
@pierreporte I currently can't add a datum which over-constraint the sketch, it's removed. Can you?

pierreporte, 2024-09-01:
I also can’t add a redundant dimension (other constrains I can) but apparently it’s possible because I’ve seen some screencast earlier. In any case, I think it would be better to make or keep the sketch overconstrained than trying to be clever.

Mannshoch, 2025-04-29:
May add a way to see which constrains are part of a constrain problem. Sometimes you add a constrain that cause an error but you are only sure that this constrain is correct. It would be helpful, if you could highlight what constrains are part of the Problem and may even experiment with temporary deactivate this or that constrain.