FreeCAD/FreeCAD #14153: FreeCAD Logo Conveys and Unprofessional Identity for the Project.

obelisk79, 2024-05-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Organizational identity is critical in modern society. Particularly in the tech industry. The current FreeCAD Logo, has a unique identity but suffers from poor general design. The leads towards an immediate sense of unprofessional/amateurish perception of the project as it is generally the first identifiable aspect of the project a user will encounter.

### Full version info

```shell
All Versions
All Operating Systems
```

### Subproject(s) affected?

None

### Anything else?

It is recommended to poll the community at-large for both submissions and a vote in order to allow the project management (ie maintainers group) select from a pool of the top 5 most popular community submissions. The voting of the maintainers should be conducted in private in the even they determine an option other than the 'popular' choice is selected.

The resulting selection will be submitted to github via an initial PR and contributing volunteers will begin a systematic update of all instances of the logo throughout the codebase, website, and documentation.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GAZ082, 2024-05-20:
We already vote. The issue is, the candidates are not good enough. I say lets pour some euros, organize a contest where we convey pro designers.

Also, the logo is just part of the puzzle. I'd put into the contest submission a splash screen and even a UI mockup to use as an unifying guideline for the UX/UI incremental improvements.

katepavlu, 2024-05-20:
Let's not make perfect be the enemy of good. The 5 top candidates may not be platonic ideals of good, but they are all very nicely designed and they all look more professional than the current logo. Throwing them out because something better might come at some indeterminate point in the future is not very productive. Like was said, incremental improvement.

GAZ082, 2024-05-20:
@katepavlu my proposal is not for now, as it will require time and resources. Now we should keep using the current process, which is fun and allows community participation. But as obelisk put well, the general feel is that the output of the process does not have a high standard.

FC is a project that attracts engineers, so we have a lot of novel solutions in that space, but this is not Inkscape that attracts designers.

MisterMakerNL, 2024-05-20:
@GAZ082 It's not about being the best, it's about making it better then it was. When we chose a new logo we can improve on it in the upcoming years to come.

maxwxyz, 2024-05-20:
definitely needed change and there are some good looking ones under the top 5! There were so many votes so I guess the majority won't be choosing a bad looking one out of the 50 entries.

cad-maker, 2024-05-21:
I support the work of the Design Working Group (DWG). I assume they are the drivers for this. I’m also passionate about FreeCAD. Which compells me to send my honest feedback as a community member. Please take it as intended: constructive criticism, food for thought for future initiatives.

- This issue states "the current logo suffers from poor general design". This is the whole point of the rebranding. Yet it does not expand on that problem statement. 
- This issue was created after the logo contest was finished. The other way around would have made more sense. First state the problem, then work out a solution.
- The intention is for the FreeCAD brand to convey a more professional, less amateurish look. A contest is probably the most unpredictable, unstructured and less professional way of solving this. There is no mechanism to discuss the desired outcome, to do peer review, or to even iterate on a design.
- Design deserves the same respect as code. Just imagine if the maintainers discussing how to solve the Topological Naming Problem had decided the best way to tackle it was a contest. Instead of 1) planning, 2) implementation with peer-reviewed, iterative changes.
- A contest overrides another pillar of Open Source: collaboration.
- The whole contest felt rushed. It’s understandable that a new logo is wanted for FreeCAD 1.0. But it seems like a decision that will lock the logo to a state of not being quite there for the next few years.
- Anyone could submit a vote, there was no quality control or provision to mitigate unfair voting. If someone was to create a bot or grab a bunch of friends to submit 1000 votes for their favorite logo, it would have made it to the top 5.
- The contest organizer was also a participant of the contest. Their logo has also been chosen as part of the top 5. I do not think there was ill intent. I also think only good intentions were at play here. It just poses an ethical consideration that wasn't discussed.
- Some of the current maintainers were the same people who chose (created?) the current logo. They will be the ultimate decision makers for the new logo now, from the top 5 contest winners.
- This is a subjective opinion. The contest logos, which will supposedly offer a less amateurish perception… look very amateurish. From a design POV, it’s concerning that seemingly the DWG have not even considered that. With full respect to anyone who took the time to create a contest entry, I'm of the same [opinion of that person in Reddit](https://www.reddit.com/r/FreeCAD/comments/1bm0dqz/comment/kwbzcm2/).

yorikvanhavre, 2024-05-21:
> Some of the current maintainers were the same people who chose (created?) the current logo. They will be the ultimate decision makers for the new logo now, from the top 5 contest winners.

I, for one, am abstaining from voting on the new logo.

obelisk79, 2024-05-21:
>     * This issue states "the current logo suffers from poor general design". This is the whole point of the rebranding. Yet it does not expand on that problem statement.

This whole effort was never really supposed to be a 'rebranding' this is a misnomer brought about by members of the community. I stated this was supposed to be a refresh/evolution of the existing logo.
 
>     * This issue was created after the logo contest was finished. The other way around would have made more sense. First state the problem, then work out a solution.

That was more to have a reference for compliance with the contributing guidelines before whichever logo is selected can be merged into the codebase.

> * The intention is for the FreeCAD brand to convey a more professional, less amateurish look. A contest is probably the most unpredictable, unstructured and less professional way of solving this. [...]

This isn't an organization with the funding to pay a professional designer and iterate designs, let alone have the strong organizational management structure to support such an approach.

> * Design deserves the same respect as code. Just imagine if the maintainers discussing how to solve the Topological Naming Problem had decided the best way to tackle it was a contest. Instead of 1) [...]

You see this as an override to collaboration, yet the entire effort has been community driven which in itself _is_ a form a collaboration. Community efforts like this have been pillars of FreeCAD's approach to open source. The final deliberations have been left to the core maintainer group.
 
>     * The whole contest felt rushed. It’s understandable that a new logo is wanted for FreeCAD 1.0. But it seems like a decision that will lock the logo to a state of not being quite there for the next few years.

This sounds like a similar argument of letting the pursuit of perfection be the enemy of progress.

>     * Anyone could submit a vote, there was no quality control or provision to mitigate unfair voting. If someone was to create a bot or grab a bunch of friends to submit 1000 votes for their favorite logo, it would have made it to the top 5.

This is why ranked choice voting was used, also why a top5 was passed on to the maintainers rather than just making this a popular vote gets the prize approach. As the person who had direct access to the data I saw a few weak attempts to do exactly that, and they failed.
 
>     * The contest organizer was also a participant of the contest. Their logo has also been chosen as part of the top 5. I do not think there was ill intent. I also think only good intentions were at play here. It just poses an ethical consideration that wasn't discussed.

I'd like to better understand the ethical concern here? Is this a question about my integrity?

>     * This is a subjective opinion. The contest logos, which will supposedly offer a less amateurish perception… look very amateurish. From a design POV, it’s concerning that seemingly the DWG have not even considered that.

You are, of course, entitled to such an opinion. Many factors were considered, I'm happy that there are at least a few people who work in graphic design around FreeCAD. Thanks for for sharing your thoughts.

tobiasfalk, 2024-05-21:
I think all of the 5 are good candidates of V 1.0 and a V2.0 of FreeCAD will come, and with it most likely the need/motion for a new Logo, simply to show a move forwards.
In my opinion saying something is not Professional is a very complicated and misleading.
The motion of somthing looking "Professional" is very much driven by what big and leading figures, but also something that can vary from Person to Person.

microbearsmb, 2024-05-21:
Agreed - and the decision is actually made now! Number 1 (originally 41) was chosen.

MisterMakerNL, 2024-05-27:
[image]

maxwxyz, 2024-06-06:
I'll close this.

closed 2024-06-06 by maxwxyz