FreeCAD/FreeCAD #14155: Manage project doesn´t work

gianlucafiore, 2024-05-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When you try to start the project manager, it doesn't take any action or open any windows

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git)
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-20:
an confirm: 
[image]
@yorikvanhavre FYI

FEA-eng, 2024-05-20:
It's a tiny issue but can you fix the apostrophe in the title (doesn´t --> doesn't) so that it's easier to search for this report?

maxwxyz, 2024-07-07:
@yorikvanhavre it did not seem to fix the bug:
[image]

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.3
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.21

```

yorikvanhavre, 2024-07-08:
Oops apparently I committed a file that shouldn't have been... PR coming

closed 2024-07-11 by yorikvanhavre