FreeCAD/FreeCAD #14165: WB TabBar: Scrolling between tabs loads every WB

maxwxyz, 2024-05-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Previously you could scroll through the TabBar WB selector to see all available WBs and change them easily.
After fixing https://github.com/FreeCAD/FreeCAD/issues/13729 it is not possible anymore.
Scrolling loads every WB so it has a huge delay and is not ideal. Also it is not possible to quickly sroll to the end to add a new one as you have to load every WB in between. A workaround is to click on the tiny arrows very often to get to the end or the preferred WB you are looking for.
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git)
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.13
  * Curves 0.6.36
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

microbearsmb, 2024-05-22:
Don't think it's in doubt, but I can replicate this behavior.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git)
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.20
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * ToolbarStyle

```

This issue existed back when I used the TabBar addon, before this feature was merged. Because of it, I've never pinned enough workbenches to necessitate scrolling. If this is solved, it would save quite a bit of headache using this (otherwise very convenient) feature.

kadet1090, 2024-06-09:
The best thing I can do is to disable the scroll wheel support as I don't see any way of simply scrolling the tabbar because Qt does not expose me anything. I'd have basically to reimplement the whole widget which won't be possible until 1.0.

closed 2024-06-24 by chennes