FreeCAD/FreeCAD #14166: Addon Manager: Allow to "focus" on a single addon

pieterhijma, 2024-05-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, at least to the best of my knowledge, it is not possible to open the Addon Manager to show a specific addon and its update status. 

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37369 (Git)
Build type: DEBUG
Branch: main
Hash: f814ef63aa6dce9067b507da3fc34c8cc017d882
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk , OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenTheme 2024.5.16
```

### Subproject(s) affected?

Addon Manager

### Anything else?

I will submit a small PR that shows a possible solution and explains the goal as well.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-20:
You mean like deep linking to directly open a details page for a specific addon?

pieterhijma, 2024-05-20:
There is a video in #14168 that shows the requested functionality.

closed 2024-05-27 by chennes