FreeCAD/FreeCAD #14185: Assembly: Joint flipping during dragging

PaddleStroke, 2024-05-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://github.com/FreeCAD/FreeCAD/assets/19984177/3fd2bceb-34da-4ecb-8751-ae5b4fe64859

Such 'crazy moves' can happen quite easily.

I think it should be possible to add a check to ignore drag steps if a joint has flipped.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

drwho495, 2024-05-21:
So, are you saying to stop dragging if a inversion is detected? There could be ways to continue the fix and revert back, like you save the position/rotations of each joint every update and then go back down the list until it is fixed.

PaddleStroke, 2024-05-21:
No, just ignore the step. So if a solution is bad then it is just ignored.

drwho495, 2024-05-21:
oh ok, that will work too.

PaddleStroke, 2024-05-31:
@drwho495 I'm working on a 'solve validation' that will check for various things, including gounded parts moving. It will reject the positions returned by the solve if the validation does not pass.

@chennes Is there a way for me to assign myself to issues? It would help me keep track of issues that I need to tackle.

chennes, 2024-05-31:
> @chennes Is there a way for me to assign myself to issues? It would help me keep track of issues that I need to tackle.

I don't know, but you can always ask me or @maxwxyz to do it.

PaddleStroke, 2024-06-01:
> but you can always ask me or @maxwxyz to do it.

I thought about that but I guess it can get very tiring for you guys

PaddleStroke, 2024-07-22:
I will close this issue now as it seems joint flipping is not happening anymore since @aiksiongkoh updated the solver.
Dragging is not perfect but it does not explode as in the video.

closed 2024-07-22 by PaddleStroke