FreeCAD/FreeCAD #14190: Part should have the Create Sketch button

FEA-eng, 2024-05-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I know that toolbars can be customized and sketches are mostly created in Part Design but I think that it would be nice to have a button for a new sketch in Part wb as well. After all, operations like extrusion need some input and it's usually a sketch. So it would be convenient to have such a shortcut available by default.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git)
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mno13, 2024-05-21:
Sketch and also datum creation tools (axis, line, plane). Also in other workbenches these tools should be available - especially in Sheet Metal and Assembly.

FEA-eng, 2024-05-21:
> Sketch and also datum creation tools (axis, line, plane). Also in other workbenches these tools should be available - especially in Sheet Metal and Assembly.

I would definitely appreciate datums outside of Part Design but the problem is that they need a Body at the moment. Sketches, on the other hand, can already be created outside of the PD wb so it's just a matter of making a shortcut.

PR #14318 (2024-06-10): Part: Add Sketcher_NewSketch to Part toolbar 
fixes #14190

adds Sketcher_NewSketch to the Part menu and toolbar (right before Extrude)

[image]

[image]
closed 2024-06-10 by chennes