FreeCAD/FreeCAD #14193: FEM: Rigid body constraint input fields are not deactivated on Windows

FEA-eng, 2024-05-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Input fields of the rigid body constraint are working properly on both Windows and Ubuntu but I've noticed a difference in the way they can be accessed. On Ubuntu, they are deactivated (greyed out and non-editable) depending on the mode for a given DOF. On Windows, they are all active (editable) regardless of the Mode setting. It's a minor issue but it would be nice if the input fields could be deactivated this way also on Windows to reduce the possible confusion when a value is entered but isn't working because the mode is still Free.

Windows:

[image]

Ubuntu:
...

@marioalexis84 FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git)
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2025-07-13:
@maxwxyz Could you reopen this? I missed the warning about its stale status.

luzpaz, 2025-12-07:
@FEA-eng is this still relevant and any idea why it would be exclusively Windows related ?

FEA-eng, 2025-12-14:
> [@FEA-eng](https://github.com/FEA-eng) is this still relevant and any idea why it would be exclusively Windows related ?

Actually, it seems to be ok on Windows now, but I've got no idea what could have fixed it. Maybe someone should have a look at it just in case.

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251209 (Git shallow)
Build date: 2025/12/09 13:53:17
Build type: Release
Branch: grafted,grafted
Hash: 5ad6a1ba58c665c23f1d217c23e74379d8c1a5a4
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

maxwxyz, 2025-12-14:
closing then

closed 2025-12-14 by maxwxyz