FreeCAD/FreeCAD #14194: Pre selecting an edge for fillet reports "Invalid edge link"

macdroid53, 2024-05-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Create new Part Design geometry, select one or more edge to be fillet-ed. Select the fillet dressup tool and one of the edges is reported as "Invalid edge link..."

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37439 (Git)
Build type: Release
Branch: main
Hash: 202aa359b3476a9e1905be8328c05e54482fe231
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * MnesarcoUtils 0.2.5
  * lattice2 1.0.0
  * Curves 0.6.36
  * ArchTextures
  * Nodes 0.1.36
  * OpenTheme 2024.5.16
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * freecad.gears 1.2.0
  * sheetmetal 0.4.14
  * Silk 0.1.5
  * Freecad-Built-in-themes-beta 1.2.2
  * kicadStepUpMod 11.1.5
  * DynamicData 2.62.0
  * workfeature-macro
  * fasteners 0.5.21
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-21:
I cannot reproduce, please add an example file or steps how to reproduce.

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git)
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.13
  * Curves 0.6.36
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14
```

macdroid53, 2024-05-21:
This was confirmed by @chennes.

Any geometry will exhibit the issue. No file required.

howie-j, 2024-05-21:
Can confirm, both fillet and chamfer is broken in current main.
Preselection -> tool or tool -> selection workflow does not matter.

[Screencast from 2024-05-21 21-13-59.webm](https://github.com/FreeCAD/FreeCAD/assets/82042580/47512568-c43c-48d1-a6c5-04251369b5c3)

maxwxyz, 2024-05-21:
I can confirm too now, butonly on shapes created with sketch extrusions.
It works for additive primitives which I've used for testing. Can you confirm this too?
[image]

howie-j, 2024-05-21:
Fillet does not work on solids created by:
- Pad
- Revolution
- Additive Loft

Does work on:
- Pipe
- Helix
- Primitives

```
OS: Fedora Linux 38 (Container Image) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37439 (Git)
Build type: Release
Branch: main
Hash: 202aa359b3476a9e1905be8328c05e54482fe231
Python 3.11.6, Qt 5.15.11, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * OpenTheme 2024.5.3
```

macdroid53, 2024-05-21:
Hmm...I just confirmed on a model that was a combined Body and Part Boolean.

I have not tried on primitives.

maxwxyz, 2024-05-21:
@bgbsww FYI

JohnOCFII, 2024-05-21:
Interestingly, for me, if I create a **new** project (PartDesign --> Sketch --> Pad --> Fillet) it fails as shown above from the other issues.  If I attempt to modify an _existing_ project, it works fine.

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git)
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.5.16
```

chopsywa, 2024-05-22:
I have found that I can make a single fillet/chamfer from a sketch based pad and this can contain multiple edges, but only after closing and reopening the file. I can add multiple consecutive fillets and chamfers if I save, close and reopen the file each time.

wwmayer, 2024-05-22:
> I can confirm too now, but only on shapes created with sketch extrusions.

No. It also fails if you create a combined object of two PD primitives, e.g. box + cylinder.

Btw, the error message is:
```
<Exception> FeatureDressUp.cpp(209): std::vector<TopoShape> PartDesign::DressUp::getContinuousEdges(const PartDesign::TopoShape &) -- Invalid edge link: Edge8
Fillet: Invalid edge link: Edge8
```

wwmayer, 2024-05-22:
In case it helps:
The passed string to getSubShape is `;Edge7;:H8ca,E.Edge8`. As a result the function TopoShape::shapeType fails to determine the correct `TopAbs_ShapeEnum` and returns `TopAbs_SHAPE` instead of `TopAbs_EDGE`. And because of this the calling instance `TopoShape::shapeTypeAndIndex` returns the invalid index 0.

If only a PD box is created and an edge selected for filleting then the passed string is e.g. `Edge7`.

wwmayer, 2024-05-22:
As said by others things work if a project is saved and reloaded. When doing so it works here, too. In this case the passed string is `Edge8`. So, the problem is that in the failing case `;Edge7;:H8ca,E.Edge8` is passed when it should be `Edge8`.

bgbsww, 2024-05-22:
Thanks.  The underlying issue here appears to be a failure to call oldElementName() on the string, so that it is split at the '.' and the Edge8 is used.  There are about 20 cases of this, and we don't have all of them.  I'm working on it.

drwho495, 2024-05-22:
I have a macro to restart your document, so you can run Ctrl-Shift-1 to restart. It is a patch, but it works.

drwho495, 2024-05-24:
> A build of bgbsww-toponamingFixShapeLookup at 0ada4351 fixed the case I ran into.

That's interesting

commit db48a78d (2024-05-24): Toponaming used mapped index to find shape.  fix #14194
closed 2024-05-24 by chennes