FreeCAD/FreeCAD #14195: X3D export on a Mac (intel) crashes application

seaniepie, 2024-05-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

At the end of export of an X3D file (from an STP on an Intel Mac) the application closes down without having created the output file. It seems to at least begin the process and get some way into it as with a particularly large STP file it sits for a while before crashing, but with a smaller file it is near immediate that it closes down. I'm happy to provide example files if needed.

### Full version info

```shell
[code]
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-12:
@seaniepie ping

luzpaz, 2024-07-18:
@seaniepie :point_up_2:

luzpaz, 2024-12-13:
@seaniepie ping

depthoffocus, 2025-01-08:
I just encountered this in 1.0.0 on an intel Mac running Sonoma (macOS 14.7.2)

Select the (only) body to export.  This file will export fine as `WebGL (*.html)`, but FreeCAD will crash if you try to export as `WebGL/X3d (*.xhtml)`

No file is generated.

[wing.FCStd.zip](https://github.com/user-attachments/files/18341149/wing.FCStd.zip)

```
OS: macOS 14.7.2
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.35
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * Curves 0.6.53
```

Relevant bit of the crash log, I think:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [63139]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-01-08 02:55:32.9510 +0000
OS Version:            macOS 14.7.2 (23H311)
Report Version:        12
Bridge OS Version:     9.2 (22P2093)
Anonymous UUID:        5FF5BD02-78A8-4188-0EAB-EF410C648824

Sleep/Wake UUID:       F9BA449D-A083-4EF3-936E-9F96D3EFB3B5

Time Awake Since Boot: 2100000 seconds
Time Since Wake:       731078 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Codes:       0x0000000000000001, 0x0000000000000010

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [63139]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 4444200944
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      108e52000-108e72000    [  128K] r-x/r-x SM=COW  /Applications/FreeCAD.app/Contents/Resources/bin/freecad

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libCoin.80.0.3.dylib          	       0x10e950560 SoCallbackAction::~SoCallbackAction() + 80
1   libCoin.80.0.3.dylib          	       0x10e9713ac SoToVRML2ActionP::~SoToVRML2ActionP() + 284
2   libCoin.80.0.3.dylib          	       0x10e96f568 SoToVRML2Action::~SoToVRML2Action() + 40
3   libFreeCADGui.dylib           	       0x10bc423d2 Gui::SoFCDB::writeToX3D(SoNode*, bool, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>&) + 914
4   libFreeCADGui.dylib           	       0x10bc43a35 Gui::SoFCDB::writeToX3DOM(SoNode*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>&) + 69
5   libFreeCADGui.dylib           	       0x10bc440aa Gui::SoFCDB::writeToFile(SoNode*, char const*, bool) + 362
6   libFreeCADGui.dylib           	       0x10b9068f4 Gui::Application::sExport(_object*, _object*) + 1412
7   libpython3.11.dylib           	       0x109af7a8e cfunction_call + 94
8   libpython3.11.dylib           	       0x109beba7e _PyEval_EvalFrameDefault + 225566
9   libpython3.11.dylib           	       0x109bb1f96 _PyEval_Vector + 4918
10  libpython3.11.dylib           	       0x109bb0be9 PyEval_EvalCode + 249
11  libpython3.11.dylib           	       0x109c5c948 run_mod + 168
12  libpython3.11.dylib           	       0x109c5ee68 PyRun_StringFlags + 104
13  libFreeCADBase.dylib          	       0x1096fab5f Base::InterpreterSingleton::runString(char const*) + 111
14  libFreeCADGui.dylib           	       0x10b9df0e6 Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) + 150
15  libFreeCADGui.dylib           	       0x10b8d4a5c Gui::Application::exportTo(char const*, char const*, char const*) + 1580
16  libFreeCADGui.dylib           	       0x10b9f292f StdCmdExport::activated(int) + 2751
17  libFreeCADGui.dylib           	       0x10b9ddd81 Gui::Command::_invoke(int, bool) + 673
18  libFreeCADGui.dylib           	       0x10b9ddaa5 Gui::Command::invoke(int, Gui::Command::TriggerSource) + 341
19  libQt5Core.5.15.15.dylib      	       0x10de812ae 0x10dd34000 + 1364654
20  libQt5Widgets.5.15.15.dylib   	       0x10acad0d1 QAction::triggered(bool) + 55
21  ???                           	    0x7fc71ca0bcf0 ???
```

depthoffocus, 2025-01-08:
The same crash occurs in 1.1.0dev-39786:

```
OS: macOS 14.7.2
Architecture: x86_64
Version: 1.1.0dev.39786 (Git) Conda
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.35
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * Curves 0.6.53
```

Relevant bit of crash log:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [63240]
Path:                  /Applications/FreeCAD 1.1.0-39786.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-01-08 03:04:15.7851 +0000
OS Version:            macOS 14.7.2 (23H311)
Report Version:        12
Bridge OS Version:     9.2 (22P2093)
Anonymous UUID:        5FF5BD02-78A8-4188-0EAB-EF410C648824

Sleep/Wake UUID:       F9BA449D-A083-4EF3-936E-9F96D3EFB3B5

Time Awake Since Boot: 2100000 seconds
Time Since Wake:       731601 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Codes:       0x0000000000000001, 0x0000000000000010

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [63240]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 4393697264
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      105e28000-105e48000    [  128K] r-x/r-x SM=COW  /Applications/FreeCAD 1.1.0-39786.app/Contents/Resources/bin/freecad

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libCoin.80.0.3.dylib          	       0x10b8bc560 SoCallbackAction::~SoCallbackAction() + 80
1   libCoin.80.0.3.dylib          	       0x10b8dd3ac SoToVRML2ActionP::~SoToVRML2ActionP() + 284
2   libCoin.80.0.3.dylib          	       0x10b8db568 SoToVRML2Action::~SoToVRML2Action() + 40
3   libFreeCADGui.dylib           	       0x108c18f62 Gui::SoFCDB::writeToX3D(SoNode*, bool, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>&) + 914
4   libFreeCADGui.dylib           	       0x108c1a5c5 Gui::SoFCDB::writeToX3DOM(SoNode*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>&) + 69
5   libFreeCADGui.dylib           	       0x108c1ac3a Gui::SoFCDB::writeToFile(SoNode*, char const*, bool) + 362
6   libFreeCADGui.dylib           	       0x1088cd094 Gui::ApplicationPy::sExport(_object*, _object*) + 1444
7   libpython3.11.dylib           	       0x106ae2c9f cfunction_call + 79
8   libpython3.11.dylib           	       0x106bd5d6d _PyEval_EvalFrameDefault + 228941
9   libpython3.11.dylib           	       0x106b9b7e4 _PyEval_Vector + 484
10  libpython3.11.dylib           	       0x106b9b589 PyEval_EvalCode + 249
11  libpython3.11.dylib           	       0x106c490d4 run_mod + 132
12  libpython3.11.dylib           	       0x106c4b5d8 PyRun_StringFlags + 104
13  libFreeCADBase.dylib          	       0x1066f537f Base::InterpreterSingleton::runString(char const*) + 111
14  libFreeCADGui.dylib           	       0x1089a78f6 Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) + 150
15  libFreeCADGui.dylib           	       0x10889b06c Gui::Application::exportTo(char const*, char const*, char const*) + 1580
16  libFreeCADGui.dylib           	       0x1089bb42f StdCmdExport::activated(int) + 2751
17  libFreeCADGui.dylib           	       0x1089a6591 Gui::Command::_invoke(int, bool) + 673
18  libFreeCADGui.dylib           	       0x1089a62b5 Gui::Command::invoke(int, Gui::Command::TriggerSource) + 341
19  libQt5Core.5.15.15.dylib      	       0x10ae0c2ae 0x10acbf000 + 1364654
20  libQt5Widgets.5.15.15.dylib   	       0x107c720d1 QAction::triggered(bool) + 55
21  ???                           	    0x7faf1074ce80 ???

```

JohnOCFII, 2025-01-08:
Can confirm that this export also fails for me.  M1/Arm Mac.  Other exports (STL, etc.) are fine.

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               FreeCAD [34712]
Path:                  /Users/USER/Documents/*/FreeCAD
Identifier:            FreeCAD
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [34703]
Responsible:           applet [34702]
User ID:               501

Date/Time:             2025-01-08 13:00:20.1920 -0600
OS Version:            macOS 15.2 (24C101)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       0A8C61DF-7671-48DF-83BD-6442ED9C53BA

Time Awake Since Boot: 210000 seconds
Time Since Wake:       185 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Codes:       0x0000000000000001, 0x0000000000000010

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [34712]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 4302995440
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1007a8000-1007c0000    [   96K] r-x/r-x SM=COW  /Users/USER/Documents/*/FreeCAD

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libCoin.80.0.3.dylib          	       0x106f81eac SoCallbackAction::~SoCallbackAction() + 68
1   libCoin.80.0.3.dylib          	       0x106f9eea0 SoToVRML2ActionP::~SoToVRML2ActionP() + 260
2   libCoin.80.0.3.dylib          	       0x106f9d56c SoToVRML2Action::~SoToVRML2Action() + 40
3   libFreeCADGui.dylib           	       0x10362bcf0 Gui::SoFCDB::writeToX3D(SoNode*, bool, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>&) + 848 (SoFCDB.cpp:469)
4   libFreeCADGui.dylib           	       0x10362d150 Gui::SoFCDB::writeToX3DOM(SoNode*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>&) + 68 (SoFCDB.cpp:624)
5   libFreeCADGui.dylib           	       0x10362d790 Gui::SoFCDB::writeToFile(SoNode*, char const*, bool) + 324 (SoFCDB.cpp:685)
6   libFreeCADGui.dylib           	       0x1032f1a60 Gui::ApplicationPy::sExport(_object*, _object*) + 1180 (ApplicationPy.cpp:762)
7   libpython3.13.dylib           	       0x1021c29a4 cfunction_call + 352
8   libpython3.13.dylib           	       0x10215836c _PyObject_MakeTpCall + 340
9   libpython3.13.dylib           	       0x1022c8514 _PyEval_EvalFrameDefault + 10344
10  libpython3.13.dylib           	       0x1022c4ce8 PyEval_EvalCode + 276
11  libpython3.13.dylib           	       0x1023508b4 run_eval_code_obj + 172
12  libpython3.13.dylib           	       0x1023509c0 run_mod + 220
13  libpython3.13.dylib           	       0x102353f2c PyRun_StringFlags + 120
14  libFreeCADBase.dylib          	       0x100bc10a4 Base::InterpreterSingleton::runString(char const*) + 108 (Interpreter.cpp:249)
15  libFreeCADGui.dylib           	       0x1033ca308 Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) + 168 (Command.cpp:725)
16  libFreeCADGui.dylib           	       0x1032ac670 Gui::Application::exportTo(char const*, char const*, char const*) + 1376 (Application.cpp:824)
17  libFreeCADGui.dylib           	       0x1033dbbac StdCmdExport::activated(int) + 2404 (CommandDoc.cpp:534)
18  libFreeCADGui.dylib           	       0x1033c90d8 Gui::Command::_invoke(int, bool) + 520 (Command.cpp:444)
19  libFreeCADGui.dylib           	       0x1033c8e94 Gui::Command::invoke(int, Gui::Command::TriggerSource) + 348 (Command.cpp:408)
20  libQt6Core.6.7.3.dylib        	       0x106b03c58 void doActivate<false>(QObject*, int, void**) + 1376
21  libQt6Gui.6.7.3.dylib         	       0x1061d308c QAction::activate(QAction::ActionEvent) + 284
22  libQt6Core.6.7.3.dylib        	       0x106afb93c QObject::event(QEvent*) + 644
23  libQt6Widgets.6.7.3.dylib     	       0x105db30a0 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 272
24  libQt6Widgets.6.7.3.dylib     	       0x105db4004 QApplication::notify(QObject*, QEvent*) + 508
25  libFreeCADGui.dylib           	       0x1033838a0 Gui::GUIApplication::notify(QObject*, QEvent*) + 192 (GuiApplication.cpp:93)
26  libQt6Core.6.7.3.dylib        	       0x106baed50 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
27  libQt6Core.6.7.3.dylib        	       0x106bb04c0 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 512
28  libqcocoa.dylib               	       0x108150a1c QCocoaEventDispatcherPrivate::processPostedEvents() + 140
29  libqcocoa.dylib               	       0x108151b20 QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void*) + 436
30  CoreFoundation                	       0x18dbb8894 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 28
31  CoreFoundation                	       0x18dbb8828 __CFRunLoopDoSource0 + 176
32  CoreFoundation                	       0x18dbb858c __CFRunLoopDoSources0 + 244
33  CoreFoundation                	       0x18dbb7128 __CFRunLoopRun + 840
34  CoreFoundation                	       0x18dbb6724 CFRunLoopRunSpecific + 588
[... cut ...]
44  libFreeCADGui.dylib           	       0x1032b3a48 (anonymous namespace)::tryRunEventLoop(Gui::GUISingleApplication&) + 484 (Application.cpp:2206) [inlined]
45  libFreeCADGui.dylib           	       0x1032b3a48 (anonymous namespace)::runEventLoop(Gui::GUISingleApplication&) + 484 (Application.cpp:2235) [inlined]
46  libFreeCADGui.dylib           	       0x1032b3a48 Gui::Application::runApplication() + 1916 (Application.cpp:2315)
47  FreeCAD                       	       0x1007af8d4 main + 3120 (MainGui.cpp:295)
48  dyld                          	       0x18d750274 start + 2840
```

xtemp09, 2025-01-17:
I cannot reproduce the crash.

```ruby
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39863 (Git)
Build type: Debug
Branch: HMI
Hash: 0a3d41b7db1d4a776232fd7409ce3f63c997c8b8
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

depthoffocus, 2025-01-17:
> I cannot reproduce the crash.

I'm sure I've had it working on Windows before myself, too -- this appears to be a Mac issue.

depthoffocus, 2025-09-27:
Still relevant -- still crashes if you export as WebGL/X3D (*.xhtml), X3D (*.x3d), or VRML (e.g. *.vrml)

```
OS: macOS 15.7
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: OpenCascade/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.75/2
Installed mods: 
  * ThreadProfile 1.97.0
  * freecad.gears 1.3.0
  * OpenTheme 2025.5.20
  * fasteners 0.5.42
  * lattice2 1.0.0
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.55
  * Curves 0.6.70
```