FreeCAD/FreeCAD #14196: Draft: Select Plane

gianlucafiore, 2024-05-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

### When you try to select a work plane that is automatically referenced to a face, it generates the following error
[image]

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git)
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

gianlucafiore, 2024-05-21:
[image]

### I have just realized that the error comes from when I want to use a face of a volume generated by extrusion as a work plane, and not for example on a wall of the BIM tool

[image]

gianlucafiore, 2024-05-21:
[image]

**Another mistake I just detected is that when selecting several faces from one or more volumes obtained by means of an extruded plane, only the first selected face is illuminated, but the second one is not illuminated by holding ctrl. Pressed**

maxwxyz, 2024-05-21:
I cannot reproduce this. Can you add a demo file?

Roy-043, 2024-05-21:
Looking at the error message (2nd image) this may be related to toponaming code.

kaiwas, 2024-05-22:
> Looking at the error message (2nd image) this may be related to toponaming code.

This problem was relevant even before yesterday's update.

yorikvanhavre, 2024-05-22:
I can't reproduce the problem, nor with the provided file or a clean new shape... But reading the error message, it seems to indicate an edge and a face were selected? Although that too shouldn't produce an error, but just indicate a message

Could you provide a precise step-by-step instruction on how to reproduce the error?

gianlucafiore, 2024-05-22:
1- I create some volumes from planes (rectangles, polylines), extruded. And I also create a wall with some minimal complexity to compare

2- the first error registered is that I want to establish the personalized work plane on any face of said volumes and I get the error detailed above.  Not happening like this when I select a plane of a wall type volume

3- the next error that I identify is when, in the volumes (generated from extruded planes) I want to select by pressing the ctrl key.  several faces of the same or different volumes, or faces and edges, or several edges, etc. It does not do so, only the first selected face/edge/vertex and the following ones are illuminated, although it does show in the tree on the left that all the different objects are being selected

yorikvanhavre, 2024-05-22:
Okay I see now. I still don't get an error, though, but indeed doing this:

1) Create a Draft Rectangle
2) Extrude it using Part Extrude
3) Cannot place the working plane on any of its faces anymore (no error though, the WP gets placed on the current view for me)

The selection looks OK though (Gui.Selection.getSelectionEx()[0].SubElementNames returns correct things like [Face1]

Investigating more...

yorikvanhavre, 2024-05-22:
>The selection looks OK though (Gui.Selection.getSelectionEx()[0].SubElementNames returns correct things like [Face1]

Actually not always. Now I get `(';Edge2;:G;XTR;:H210:7,F.Face2',)`

Roy-043, 2024-05-22:
There is an underlying problem that needs to fixed as well I think. `SubElementNames` should not contain such a string. Maybe one of the toponaming devs can comment? @CalligaroV can you have a look?

There are other occurrences of `FreeCADGui.Selection.getSelectionEx("", 0)` in the Draft and BIM code that would need to be checked as well.

yorikvanhavre, 2024-05-22:
Indeed, in
BIM/ArchCutPlane.py
BIM/bimcommands/BimCutPlane.py
Draft/WorkingPlane.py

closed 2024-05-27 by chennes

Roy-043, 2024-06-06:
@CalligaroV Can you please give some information. PR #14206 is not the solution, it just introduces a new problem. I know that you and the other toponaming devs are busy, but without any concrete info we are sort of floundering. Thanks.

CalligaroV, 2024-06-07:
Hi @Roy-043!

Sure, I'll give it a look today and give you a feedback.

Unfortunately I haven't studied the logic of this method before so I don't know if I'll be able to find a fix in short time but I'll update you with my discoveries later today.

CalligaroV, 2024-06-07:
So, here are the checks I did.

I started reverting #14206, tried to reproduce the original error of this issue but I'm not able to reproduce it.
I guess that some modifications made in these weeks have effect also on this one.

After that I started analyzing the code related to `SubElementNames`, searching for differences between main and RealThunder's fork.
I analyzed `src/Gui/SelectionObjectPy.xml`, `src/Gui/SelectionObjectPyImp.cpp`, `src/Gui/SelectionObject.h`, `src/Gui/Selection.h` and `src/Gui/Selection.cpp`.

The code related to `SubElementNames`, `getSubElementNames` and `getSubNames` in `src/Gui/SelectionObjectPy.xml`, `src/Gui/SelectionObjectPyImp.cpp` and `src/Gui/SelectionObject.h` is the same in both codebases.

Also the code for `getSelectionEx`, `sGetSelectionEx`, `getObjectOfType` and `sAddSelection` in `src/Gui/Selection.h` is the same as well as the code for `sGetSelectionEx` and `sAddSelection` in `src/Gui/Selection.cpp`.

**Differences**:

In `src/Gui/Selection.h`:
- `_SelList`: in RT's fork isn't declared as a `mutable std::list<_SelObj>` but as a `mutable SelContainer` (custom data type defined at https://github.com/realthunder/FreeCAD/blob/0e9cf4aa7c7bac94a83edd07b4bd74a71a26446c/src/Gui/Selection.h#L861)
- `getSelectionEx`: in RT's fork the 3rd argument is `_SelList.get<0>()`. I guess it's a consequence of the different declaration of `_SelList` https://github.com/realthunder/FreeCAD/blob/0e9cf4aa7c7bac94a83edd07b4bd74a71a26446c/src/Gui/Selection.cpp#L437
- `getObjectList`: in RT's fork the 3rd argument is `T &objList`. I guess it's for the same reason of the previous point https://github.com/realthunder/FreeCAD/blob/0e9cf4aa7c7bac94a83edd07b4bd74a71a26446c/src/Gui/Selection.cpp#L369C35-L369C45
- `addSelection`: in RT's fork there's one more definition of this method that uses an `App::SubObjectT` object as argument https://github.com/realthunder/FreeCAD/blob/0e9cf4aa7c7bac94a83edd07b4bd74a71a26446c/src/Gui/Selection.h#L364C29-L364C44

In `src/Gui/Selection.cpp`:
- `getObjectList`: in RT's fork there are some lines of code more than in main https://github.com/realthunder/FreeCAD/blob/0e9cf4aa7c7bac94a83edd07b4bd74a71a26446c/src/Gui/Selection.cpp#L404
- `getObjectOfType`: in RT's fork `|| sel.elementName.second.empty()` is added to the conditional statement https://github.com/realthunder/FreeCAD/blob/0e9cf4aa7c7bac94a83edd07b4bd74a71a26446c/src/Gui/Selection.cpp#L608
- `addSelection`: in RT's fork there are the lines added here https://github.com/realthunder/FreeCAD/blob/0e9cf4aa7c7bac94a83edd07b4bd74a71a26446c/src/Gui/Selection.cpp#L1256 and here https://github.com/realthunder/FreeCAD/blob/0e9cf4aa7c7bac94a83edd07b4bd74a71a26446c/src/Gui/Selection.cpp#L1268

I imported the differences for `getObjectList` and `getObjectOfType` but, with no surprise, this didn't help to solve the `FreeCADGui.Selection.getSelectionEx("", 0)[0].SubElementNames` issue

I haven't tried the other modifications (related to the different declaration of `_SelList` and the lines added for `addSelection`) as those require importing/modifying more code.

Last check I did was trying to call `FreeCADGui.Selection.getSelectionEx("", 0)[0].SubElementNames` in RT's fork and also there the output is the same we get now in main.

**Conclusion**:

By now the only thing that comes in my mind is to modify one or more of the methods I analyzed to get the outputs that we want, but probably doing so we'll cause more issues that the ones we may solve

Roy-043, 2024-06-13:
Thanks. For the Draft WP the issue has been  resolved because `Part.getShape` can now handle the SubElementNames as reported by Yorik. I will therefore revert #14206.

Roy-043, 2024-06-13:
I have reopened #13248.

PR #14206 (2024-05-27): Draft: Enable TNP in SelectPlane
fixes #14196
commit b057e6eb (2024-05-27): Draft: Enable TPN in SelectPlane - fixes #14196