FreeCAD/FreeCAD #14197: BIM: Error when trying to place doors and windows on walls

gianlucafiore, 2024-05-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to place doors and windows on the walls, I get the following error the first time, they don't self-host, but then manually they do

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git)
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-21:
I can reproduce this. 
@yorikvanhavre FYI

yorikvanhavre, 2024-05-22:
I don't get any error message, but indeed auto hosting is disabled. I'll submit a PR

gianlucafiore, 2024-05-22:
https://github.com/FreeCAD/FreeCAD/assets/50465150/91681098-88ae-4b30-bcd7-0ebbe200adcf

I'm adding a short video of the steps described above

Roy-043, 2024-05-22:
Is this video related to this issue? I do not see previously described steps.

gianlucafiore, 2024-05-22:
> ¿Este video está relacionado con este tema? No veo los pasos descritos anteriormente.

I'm sorry I made a mistake it was for another topic

I just realize, I've already deleted it

commit d119dc57 (2024-05-22): BIM: Fixed windows autohosting - fixes #14197