FreeCAD/FreeCAD #14200: Addon Manager: update fails to overwrite files, creates nested folders instead

Roy-043, 2024-05-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When updating addons I have the issue that new files are wrongly placed in nested folders. For example after updating the Fasteners WB I have this folder:
C:\Users\Name\AppData\Roaming\FreeCAD\Mod\fasteners\FsData\FsData\

In all folders a new subfolder with the same name has been created and new files are placed in those subfolders.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git)
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-22:
@chennes fyi

Roy-043, 2024-07-30:
Problem still occurs here.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods: 
  * boltsfc 2022.11.5
  * dodo 1.0.1
  * sheetmetal 0.4.23
```

chennes, 2024-07-30:
And you are using ZIP installation, right?

Roy-043, 2024-07-31:
Correct.

chennes, 2024-08-18:
I'm going to upgrade this to a blocker, this should get fixed before 1.0 (but is OK for release candidates in the meantime).

PR #15953 (2024-08-19): Addon Manager: Fix ZIP installation
Fixes #14200 by always deleting the old version before unzipping the new one.
closed 2024-08-19 by WandererFan