FreeCAD/FreeCAD #14203: TechDraw : problem of visibility of the white cross of the new context dimension tool on a white background.

Cobras62, 2024-05-22:
I just checked and the color is indeed a preference.
```
    ParameterGrp::handle hGrp =
        App::GetApplication().GetParameterGroupByPath("User parameter:BaseApp/Preferences/View");
    color = hGrp->GetUnsigned("CursorCrosshairColor", color);
```

It is currently defined in the shared class ToolHandler, which is shared between sketcher and techdraw. So we need to change this to have separate colors. Please open an issue I'll do this later.

_Originally posted by @PaddleStroke in https://github.com/FreeCAD/FreeCAD/issues/13525#issuecomment-2124125432_

Cobras62, 2024-05-22:
It's not a problem with the whole icon, just the white cross. Alternatively, can't we do what mouse pointers do? A white cross surrounded by a black thread? That way you can see the cross in all conditions and it solves the problem :)

Reqrefusion, 2024-05-22:
[image]
Does this solve the problem? Or is something else needed?

maxwxyz, 2024-05-22:
I'd make it one color but maybe @PaddleStroke could confirm how this should be done as the preferences should be able to overwrite the right part of the logo with a color in FreeCAD itself

Cobras62, 2024-05-22:
> ![TechDraw_Dimension_Pointer](https://private-user-images.githubusercontent.com/39885728/332929768-029e717e-b94b-4713-a6ca-163ed6b83216.svg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0MDY4NTMsIm5iZiI6MTcxNjQwNjU1MywicGF0aCI6Ii8zOTg4NTcyOC8zMzI5Mjk3NjgtMDI5ZTcxN2UtYjk0Yi00NzEzLWE2Y2EtMTYzZWQ2YjgzMjE2LnN2Zz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIyVDE5MzU1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU0ZTVmMjU3Y2EyNTg2NjZkZmIwZWUzN2I5N2ZjNmFjNDY4ODZmODQyYzhjYTVhMzM1MTRiMTk1YTlhZTgwZjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Jagmj6IO2bFBIE0Zws-pCxSeA9ZaNfA62Ae2e64skS8) Does this solve the problem? Or is something else needed?

It looks really good like that!  I think this cross design would allow it to be seen in all conditions :)

PaddleStroke, 2024-05-23:
The way it is currently done is that the color of the cross (white) is replaced programatically by the ToolHandler class. Such that the color can be defined by a preference.

I think the best solution is to add a second preference color for the cross, which we would use for TD.

PaddleStroke, 2024-05-23:
Alternatively, if the proposed icon works on all background color, we can just use this and get rid of the preference.

Cobras62, 2024-05-23:
@PaddleStroke : It's worth testing. The problem is that some people work on their technical drawings on a black background, so the cross needs to be visible on a white and/or black background.

Reqrefusion, 2024-05-23:
> Alternatively, if the proposed icon works on all background color, we can just use this and get rid of the preference.

Yes it works. It's made to work in any theme for a bunch of icons. Then I add this issue to my PR.

maxwxyz, 2024-05-23:
It should be the universal icon and the color is set in the preferences.

Reqrefusion, 2024-05-23:
If this problem persists when I finish the technical drawing revision, I will add the relevant icon. I hope a universal solution can be made by then.

Cobras62, 2024-05-24:
@Reqrefusion : Is there any way I can test this new icon?

Reqrefusion, 2024-05-25:
> @Reqrefusion : Is there any way I can test this new icon?

Maybe you can test it by compiling it with the icon.

PaddleStroke, 2024-05-27:
@Reqrefusion you'll make a PR with this new icon?

Reqrefusion, 2024-05-27:
> @Reqrefusion you'll make a PR with this new icon? Or do you need me to do it?

Can this problem be solved with a simple icon change? If it can be solved, I don't need you. I have to ask because @maxwxyz is very sensitive about these issues.

PaddleStroke, 2024-05-27:
If you tested the icon and it works on both background the solution will already be good. Indeed the general mouse pointer is white with black outline and its good.

Cobras62, 2024-05-27:
I'm willing to give it a try, but I've never compiled FreeCAD ^^.
But why not try, I could learn new things :)

FEA-eng, 2024-07-01:
@maxwxyz Could you add the 1.0 label also here? Currently, it's hard to use the tool because of this.

closed 2024-07-01 by maxwxyz