FreeCAD/FreeCAD #14219: ViewObject.Selectable is broken for objects in App::Part

krushia, 2024-05-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Things in a part container (App::Part) can't be made unselectable in the 3D view.

1. Select some geometry
2. Property View > View tab > Set [Selection] Selectable to false
3. Geometry no longer selectable in 3D view **(GOOD)**
4. Create a Part using Std_Part from Structure toolbar
5. Drag and drop the "unselectable" object into the Part container (in the Tree view)
6. Geometry is selectable despite ViewObject.Selectable set to false **(BUG)**

0.21.2 suffers from this as well. 

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.36 (Disabled)
  * A2plus 0.4.64a (Disabled)
  * DynamicData 2.62.0
  * Assembly4 0.50.12 (Disabled)
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9 (Disabled)
  * fasteners 0.5.20 (Disabled)
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools (Disabled)
  * workfeature (Disabled)
  * MnesarcoUtils 0.2.5 (Disabled)
  * OpenTheme 2024.5.3
  * Manipulator 1.5.7
  * Assembly3 0.12.2 (Disabled)
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement (Disabled)
  * PieMenu 1.5.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-22:
Is the Part object also set selectable false?

krushia, 2024-05-22:
App::Part has no Selectable property in ViewObject. The object inside is selected.

krushia, 2024-05-22:
[selectable-bug.zip](https://github.com/FreeCAD/FreeCAD/files/15408443/selectable-bug.zip)

Small example made in 0.21.2 attached.

krushia, 2024-05-22:
It gets weirder. Object shapes in an App::Part can be selected when the object's ViewObject.Selectable is false, but sub-shapes (like faces) cannot be selected.

PaddleStroke, 2024-11-12:
FYI I encountered this bug and have tried to change the Selectable mechanism to use SoPickStyle instead : 

> I have just tried to replace the current system by a SoPickStyle in ViewProviderGeometricObject, and it appears that it changes nothing :
> 
> The bug with App::Part is still here

PaddleStroke, 2024-11-13:
Another bug that is very closely related to this one : https://github.com/FreeCAD/FreeCAD/issues/17834

NomAnor, 2025-03-18:
This seems to be fixed in the main branch. I can't select any sphere in the provided file which looks to be correct. The linked issue for `App::Links` seems still to exist.

```
OS: Arch Linux (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40845 (Git)
Build date: 2025/03/18 14:59:22
Build type: Debug
Branch: main
Hash: c1ada6bc4070df7804ccea69773b0cfa0d8226fd
Python 3.13.2, Qt 5.15.16, Coin 4.0.3, Vtk , OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/141.585/1
Installed mods: 
  * fasteners 0.5.34
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * 3DfindIT 1.2.0
 ```

PaddleStroke, 2025-03-19:
I can still reproduce in 40650. Perhaps it was fixed since. I will rebase an test

closed 2025-09-02 by maxwxyz