FreeCAD/FreeCAD #14221: Block constraint shows wrong DOF

chrisb-github, 2024-05-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the attached sketch the block constraint shows 2 DOF, while there should be zero, see also https://forum.freecad.org/viewtopic.php?t=87524.

[BlockOfFreedom.zip](https://github.com/FreeCAD/FreeCAD/files/15408298/BlockOfFreedom.zip)
(Save as *.fcstd instead of *.zip)

### Full version info

```shell
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36904 (Git)
Build type: Release
Branch: main
Hash: 0622cb784e199053656b50854e92eff806130408
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.2.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fasteners 0.5.15
  * ExplodedAssembly
  * Curves 0.6.29
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2025-07-17:
Still relevant. @maxwxyz Can you reopen it? @chrisb-github reported a few important Sketcher bugs but he is not active anymore so he can't prevent them from being close automatically.

PaddleStroke, 2026-09-07:
Related to https://github.com/FreeCAD/FreeCAD/pull/26791
Block constraints are kind of weird.