FreeCAD/FreeCAD #14224: BIM: Multiple errors with anotations

gianlucafiore, 2024-05-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

### Errors are generated from adding dimensions, annotation guidelines, texts, that whole series of errors arise, when I press ESC. To finish the command, multiply "Wires" in the tree on the left

[image]
[image]
[image]

### And also the button to create a new page and place views from the modeling space, don't work
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git)
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-05-23:
The above error I can't reproduce: All annotation tools work for me here...

There is a very strange PyCXX error, though. I've never seen that before, and it's not related to BIM or Draft.
Do other Draft tools (line, polyline,...) work normally?

The second error is fixed in #14230 

FYI it's easier for us if you create one separate issue for each separate problem...

gianlucafiore, 2024-05-23:
Apologies, I understand, I will take it into account for the following reports. I am going to attach a video so that it is displayed as inserting any annotation and then pressing ESC to exit or deselect generates a wire loop

https://github.com/FreeCAD/FreeCAD/assets/50465150/5bf9aadd-7f1b-4ee4-ac69-503c17519b75

yorikvanhavre, 2024-05-24:
I can't see the video?

gianlucafiore, 2024-05-24:
> There is a very strange PyCXX error, though. I've never seen that before, and it's not related to BIM or Draft.
Do other Draft tools (line, polyline,...) work normally?
> 

Yes, I haven't problems on that

yorikvanhavre, 2024-05-24:
I'm reproducing the same steps as you:

1. Click the Draft Label button
2. Press ESC - The command stops
3. Click the Draft Label button again
4. Click 3 points - The command completes

Nothing happens for me, no new wires are created, everything is normal...
Could you turn the report view on, and paste here any error that happens there?

gianlucafiore, 2024-05-27:
@yorikvanhavre 
During the weekend I downloaded the last weekly compilation, and decided to reset all the defaults of Freecad, I could notice that many errors were self-corrected by the mere fact of changing the color theme, I stopped using the Dark or Open dark that I had installed. Freecad seems to work better in general, plus I stopped having the annotation errors, or at least I couldn't reproduce them again

closed 2024-05-27 by gianlucafiore