FreeCAD/FreeCAD #14233: Dev - Sketcher - Pasted circle is arbitrarily constrained .

GAZ082, 2024-05-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I copy and when I paste a circle it gets coincident constrained on an arbitrary entity. I have auto-constrains disabled.

https://github.com/FreeCAD/FreeCAD/assets/100194/c49878fe-84e4-4444-949d-2183c1179bcd

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git) AppImage
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-23:
@PaddleStroke FYI

PaddleStroke, 2024-05-27:
It appears that the circle you copy has a point on object constraint with an axes. I think the problem is here, the axes has a geoId of -1 if I recall correctly. It is likely that I did not handle separately the case of axes geoIds.

PR #14308 (2024-05-27): Sketcher: Fix copy of geo constrained to axis. Fixes #14233
Fixes #14233
commit 6bbf61b1 (2024-05-27): Sketcher: Fix copy of geo constrained to axis. Fixes #14233
closed 2024-05-27 by chennes