FreeCAD/FreeCAD #14234: When a Point object is in a sketch all Part Design feature tools fail, but Part workbench is fine.

macdroid53, 2024-05-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the latest builds when a point is added to a sketch it is added, by default, as geometry not construction. In the past when a geometry point was in a sketch it was ignored and Pad/Pocket/etc would work. 
Now it produces the error: ` Pad: Shape must be a wire, edge or compound. Something else was supplied.`
Is is ignored in Part workbench. In both Part and Part Design it works as expected when a single point is in the sketch and used a one profile in a loft.
[image]

[PointTests.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/15416797/PointTests.FCStd.zip)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37462 (Git)
Build type: Release
Branch: main
Hash: d12635246f86947031776a42714028b236bd05ed
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * MnesarcoUtils 0.2.5
  * lattice2 1.0.0
  * Curves 0.6.37
  * ArchTextures
  * Nodes 0.1.36
  * OpenTheme 2024.5.16
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * freecad.gears 1.2.0
  * sheetmetal 0.4.14
  * Silk 0.1.5
  * kicadStepUpMod 11.1.5
  * DynamicData 2.62.0
  * workfeature-macro
  * fasteners 0.5.21
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-05-23:
@bgbsww is something like https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:PD_Revolution_bugfix maybe just a Log rather than a Warning??

macdroid53, 2024-05-23:
But a one point Pad/Pocket/Revolve it simply ignored it. The only situation where a point was considered was when it was one of the profiles of a loft.

Syres916, 2024-05-23:
> The only situation where a point was considered was when it was one of the profiles of a loft.

And that still works with this change, the shape is 'seen' as a `Wire` in this instance, if needed I'll have to compile realthunder's branch but I'm sure @bgbsww will know where the gotcha is hiding.

bgbsww, 2024-05-23:
As far as I can see, FaceMaker has always thrown this error if handed something not a wire.

Looking up the stack, we have:

Part::FaceMaker::addTopoShape FaceMaker.cpp:75
Part::TopoShape::makeElementFace TopoShapeExpansion.cpp:4587
Part::TopoShape::makeElementFace TopoShapeExpansion.cpp:4564
Part::TopoShape::makeElementFace TopoShape.h:2323
PartDesign::ProfileBased::getTopoShapeVerifiedFace FeatureSketchBased.cpp:258
PartDesign::FeatureExtrude::buildExtrusion FeatureExtrude.cpp:459
PartDesign::Pad::execute FeaturePad.cpp:74
...

getTopoShapeVerifiedFace is apparently calling FaceMakerBullseye instead of FaceMakerCheese.  That doesn't seem to be it.

FeaturePad calling buildExtrusion is also a new behavior with TNP, and it changes many things.
However, there don't appear to be any differences to LS3 that would cause this.

That takes us to makeElementFace.  Also nothing obvious.  But in LS3 the code goes in and out of makEFace without throwing anything, so it's in there somewhere ...

Syres916, 2024-05-23:
Thanks for the feedback, let me sleep on it, I'll probably be more use in the morning and have a quicker box to compile on.

bgbsww, 2024-05-23:
I'm still looking at it.  Pretty sure it's the tip of an iceberg that culminates in buildShape().

bgbsww, 2024-05-24:
I have a fix for this.  Little work to do to get it PR ready.  This fixes backwards compat, but still leaves the larger question open.

bgbsww, 2024-05-31:
Sorry, I didn't like that fix ultimately, which was probably a good thing.  There appears to be some code in LS3 in PartDesign/Gui/Utils.cpp in importExternalObjects called by PartDesign/Gui/Command.cpp in prepareProfileBased  that has a side/intentional effect of changing the contents of the Profile property to contain references to subelements.  My first take on this is that behavior in App is being affected by whether or not some code runs in Gui, and that needs fixing.  I might be wrong, but the depth of the issue is why it is taking so long to resolve cleanly.

mh-dm, 2024-05-31:
Can I suggest patching it temporarily while working on a proper fix to avoid more and more people hitting this issue and taking action (reporting, etc)?

Syres916, 2024-06-01:
@bgbsww thanks for the efforts

closed 2024-06-02 by chennes

Ely31, 2024-08-15:
I'm getting this same issue on version I just built yesterday, I think this should be reopened.

bgbsww, 2024-08-16:
Can you be more specific?  The file attached to this has the load save problem resulting in a ?Vertex1 reference to the section in the AdditiveLoft, but once this is corrected,  you can save and load and not see any errors or issues.

And the warning has changed to
PointTests#AdditiveLoft.Sections missing element reference PointTests#Sketch003 ;Vertex1;SKT;:H:4,V;SKT;:H:f,V.Vertex1 as it should.

So this looks like a broken file saved in an early 0.22 release, not a continuation of the original bug.

If you have an new example, that would be great.

Ely31, 2024-08-16:
[sketchPointBug.zip](https://github.com/user-attachments/files/16632174/sketchPointBug.zip)
Change the point floating off to the top left from construction to regular geometry and it fails with "Pad: Shape must be a wire, edge or compound. Something else was supplied."

bgbsww, 2024-08-16:
Yep, thanks!  Let's see what's happening...

bgbsww, 2024-08-22:
Ok, this is a geometry order affecting bug.   You can do lots of point object in sketch examples, including adding point then rectangle,  or rectangle then point and not trigger this.

Not sure if this has to do with the point being before other geometry but not the first geometry or something else that is triggering this, but it is a useful place to take to the debugger.

Ely31, 2024-08-22:
I think it's got to do with if the sketch produces a hole when padded.
This pads fine:
[image]
But add a hole to the topology like this and it fails:
[image]

bgbsww, 2024-08-25:
I agree with that hypothesis, and I'm now looking into it.

PR #14421 (2024-06-02): Toponaming: fix bug with points in sketches
Should fix #14234
PR #16037 (2024-08-26): Ignore standalone points in sketches when building faces
Fix #14234.