FreeCAD/FreeCAD #14236: [TechDraw]Confusing tool structuring

Reqrefusion, 2024-05-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The tools are interestingly divided into two, and in these two parts there are tools that replace each other or do the same job. This order creates confusion. It challenges the person while drawing. I have given more concrete examples below. I believe that the correction should start from these.

### Full version info

```shell
All Versions
All Operating Systems
```

### Subproject(s) affected?

Techdraw

### Anything else?

I'm currently working on technical drawing icons. I'm trying to give some consistency to these. There is a distinction made before me in technical drawing. And this is incredibly confusing. A distinction made between Extensions and Annotations. Annotations has much newer features and relatively better icons. The problem is which icon goes into which one.

For example, when entering these Annotations https://wiki.freecad.org/TechDraw_CosmeticCircle. Why doesn't https://wiki.freecad.org/TechDraw_ExtensionDrawCosmCircle enter this? 

Shouldn't Centerlines and threading in Extensions go entirely into Annotations? This is a problem related to grouping. Maybe it can be solved by changing the icon, but it should be changed where it is located in the toolbar.

Why https://wiki.freecad.org/TechDraw_ExtensionChangeLineAttributes this and https://wiki.freecad.org/TechDraw_DecorateLine this exist at the same time. They both do the same job. They just create more confusion.

There are some problems here that I cannot solve with just an icon change. For icons, this distinction is very strict. Annotations icons are blue, Extensions icons are yellow. I plan to remove this distinction in the icons and make the ones related to dimensioning yellow and the ones related to adding something to the drawing blue. However, as I said, there are things here that icon revision cannot solve. Considering all the tools, those that do the same job should be combined and the tools should be grouped again.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-05-23:
For context, the "blue" tools came first, then the "red" tools were created.   At one time there were a number of things that one set of tools did better than the other.   Over time the duplicated tools have become closer in capabilities.

Reqrefusion, 2024-05-23:
My mistake, I thought the ones with better looking icons were newer.