FreeCAD/FreeCAD #14238: [TD]"Smart" dimension tool does not support 3d dimensions

WandererFan, 2024-05-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Smart dimension tool consistently fails when attempting to create a dimension from 3d references. 

To reproduce:
- create new document
- add Part Cube
- add TechDraw page
- add view of Cube
- select view of Cube and 2 vertices on Cube (in 3d window)
- press smart dimension button

failure:
`1 TechDrawGui::QGVPage::getDrawPage                QGVPage.cpp           555 0x7fcfb39d18bc 
    (this) is 0x0
2 TechDrawGui::TechDrawHandler::getPage            TechDrawHandler.cpp   130 0x7fcfb39a5cd2 
    in viewPage->getDrawPage();  viewPage is 0x0
3 TDHandlerDimension::onSelectionChanged           CommandCreateDims.cpp 509 0x7fcfb382a97b 
4 Gui::SelectionObserver::_onSelectionChanged      Selection.cpp         152 0x7fd02a20baed `

### Full version info

```shell
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37449 +17 (Git)
Build type: debug
Branch: CosmeticReview4
Hash: 78cca4a7c294b9592539baade5859255f8543cbe
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk , OCC 7.7.1
Locale: English/Canada (en_CA)
Installed mods: 
  * BOLTSFC 2022.11.5
  * sheetmetal 0.2.63
  * fasteners 0.4.56
  * FreeCAD_Electric
  * freecad.gears 1.0.0
  * Curves 0.6.29
  * Assembly4 0.50.9
  * A2plus 0.4.64
```

### Subproject(s) affected?

Techdraw

### Anything else?

one problem is in void activateHandler(TechDrawHandler* newHandler).  If the 3d window is active, this will fail as the 3d window is not a MDIViewPage.  
`//! find the appropriate QGraphicsView (QGVPage) widget and invoke its activateHandler method
void activateHandler(TechDrawHandler* newHandler)
{
    Base::Console().Message("CCD::activateHandler()\n");

    auto tdhd = dynamic_cast<TDHandlerDimension*>(newHandler);
    if (!tdhd) {
        return;
    }
    auto dvp = tdhd->getViewPart();         // new method
    auto vpdv = dynamic_cast<ViewProviderDrawingView*>(Gui::Application::Instance->getViewProvider(dvp));
    if (!dvp ||!vpdv) {
        return;
    }

    auto vpp = vpdv->getViewProviderPage();
    if (!vpp) {
        return;
    }

    auto viewPage = vpp->getQGVPage();
    if (!viewPage) {
        return;
    }

    viewPage->activateHandler(newHandler);
}`

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-05-23:
Some potentially useful code:
`void CmdTechDrawDimension::activated(int iMsg)
{
    Q_UNUSED(iMsg);
    ReferenceVector references2d;
    ReferenceVector references3d;
    TechDraw::DrawViewPart* partFeat =
        TechDraw::getReferencesFromSelection(references2d, references3d);
    activateHandler(new TDHandlerDimension(references2d, references3d, partFeat));  // additional parameter
}

    //! version of getSelectionVector for 3d References
    // significant amounts of guesswork here.
    ReferenceVector& getSelectionVector3d(ReferenceEntry& ref)
    {
        if (!ref.is3d()) {
            return emptyVector;
        }
        
        std::string subName = ref.getSubName();
        if (subName == "") {
            return emptyVector;
        }

        std::string geomName = DrawUtil::getGeomTypeFromName(subName);
        if (geomName == "Face") {
            return selFaces;
        }
        else if (geomName == "Edge") {
            auto geometry = ref.getGeometry();
            auto edge = TopoDS::Edge(geometry);
            BRepAdaptor_Curve adapt(edge);
            if (adapt.GetType() == GeomAbs_Line) {
                return selLine;
            }
            if (adapt.GetType() == GeomAbs_Circle) {
                return selCircleArc;
            }
            if (adapt.GetType() == GeomAbs_Ellipse) {
                return selEllipseArc;
            }
            if (adapt.GetType() == GeomAbs_BSplineCurve) {
                return selSplineAndCo;
            }
        }
        else if (geomName == "Vertex") {
            return selPoints;
        }

        return emptyVector;
    }
`

maxwxyz, 2024-05-23:
@PaddleStroke FYI

PaddleStroke, 2024-05-29:
The smart dimension tool was designed only for the 2d reference workflow indeed, as I was not familiar with this 3d ref workflow.
I think this is going to be a fairly large project to add support to 3d ref.

furgo16, 2024-05-29:
If it can be detected, would it make sense in the meantime to show the user a message saying that 3D dimensions are not supported yet?

WandererFan, 2024-06-10:
3d dimension still work if you use the individual dimension tools.

WandererFan, 2025-02-19:
additional information in issue #19654 (closed).

PR #23101 (2025-09-01): TechDraw: random fixes
This PR brings in TechDraw related changes from @wwmayer's [branch](https://github.com/wwmayer/FreeCAD/tree/main-dev).

Fixes: #14238
closed 2025-09-01 by hyarion