FreeCAD/FreeCAD #14240: [Regression] Sketcher_ValidateSketch no more fixes missing coincidence

FlachyJoe, 2024-05-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Missing coincidences are found but the Fix button do nothing.

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37459 (Git)
Build type: Debug
Branch: main
Hash: 83d91d61c68ec6e95ae06545066665589f706c21
Python 3.11.9, Qt 5.15.10, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://forum.freecad.org/viewtopic.php?t=87853 (French)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-23:
Is this related to theTNP fix?

Syres916, 2024-05-23:
I can confirm for definite tomorrow but https://github.com/FreeCAD/FreeCAD/commit/37f0ad43f9d42a2007528bb42b6ac96cd01cde66 is looking a likely candidate so not the TNP enable from this week.

Syres916, 2024-05-24:
@wwmayer since the line https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Sketcher/Gui/TaskSketcherValidation.cpp#L206 was changed from `    sketchAnalyser.makeMissingPointOnPointCoincident();` the faulty coincidents are not being passed into https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Sketcher/App/SketchAnalysis.cpp#L449
Could you please review, TIA.

Syres916, 2024-05-24:
Note: I put together a rudimentary unit test for future reference if required https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Sketcher_Validation_Test

wwmayer, 2024-05-24:
Thanks for the test. I will have a look at the issue today evening or tomorrow...

wwmayer, 2024-05-26:
> Note: I put together a rudimentary unit test for future reference if required 

This unit test lacks of the function call sketch.detectMissingPointOnPointConstraints() and it doesn't help much to describe the actual problem.

Part 1 of the problem:
There are two instances of SketchAnalysis -- one is a member of SketcherValidation and one is a member of SketchObject and both don't know about each other.

Part 2 of the problem:
Calling makeMissingPointOnPointCoincident() alone doesn't work but a call of detectMissingPointOnPointConstraints() is needed beforehand. This internally stores information about missing coincidences which are fixed with makeMissingPointOnPointCoincident()

With #14304 the issue is fixed.

commit 5696ee82 (2024-05-27): Sketcher: Fixes #14240
closed 2024-05-27 by wwmayer