FreeCAD/FreeCAD #14242: Draft Facbinder produces no display, but creates a Facebinder in the tree view.

macdroid53, 2024-05-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Facebinder produces nothing in the 3D view.
(Same result if the face is a face of an imported step/iges file)

Create a Part Cube
Select a face of the cube
In Draft workbench click Facebinder
A Facebinder is shown in the tree view but nothing in the 3D view
[image]

[NoFacebinder.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/15419094/NoFacebinder.FCStd.zip)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37462 (Git)
Build type: Release
Branch: main
Hash: d12635246f86947031776a42714028b236bd05ed
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * MnesarcoUtils 0.2.5
  * lattice2 1.0.0
  * Curves 0.6.37
  * ArchTextures
  * Nodes 0.1.36
  * OpenTheme 2024.5.16
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * freecad.gears 1.2.0
  * sheetmetal 0.4.14
  * Silk 0.1.5
  * kicadStepUpMod 11.1.5
  * DynamicData 2.62.0
  * workfeature-macro
  * fasteners 0.5.21
  * Defeaturing 1.2.2
  * CurvedShapes 1.0.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-23:
@Roy-043 FYI

Roy-043, 2024-05-23:
Confirmed. This issue may be caused by #14169.

PR #14296 (2024-05-27): Draft: update fuse behavior in facebinder.py
Fixes #14242.

In the current dev fusing an empty shape with a face or solid no longer works (probably introduced with #14169). The reason for starting with an empty shape was the assumption that the end result would then always have a default placement. But that is only correct if multiple objects are fused.

In the new version of the code fusing no longer starts with an empty shape. The default placement issue is solved by always returning a compound.

Additionally the value for the Area property is now taken after offsetting and fusing the faces.
commit 7d086713 (2024-05-27): Draft: update fuse behavior in facebinder.py

Fixes #14242.

In the current dev fusing an empty shape with a face or solid no longer works (probably introduced with #14169). The reason for starting with an empty shape was the assumption that the end result would then always have a default placement. But that is only correct if multiple objects are fused.

In the new version of the code fusing no longer starts with an empty shape. The default placement issue is solved by always returning a compound.

Additionally the value for the Area property is now taken after offsetting and fusing the faces [... cut]
closed 2024-05-27 by yorikvanhavre