FreeCAD/FreeCAD #14245: on master branch of freecad there are some include missing

Tarot67, 2024-05-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

some include are missing in file 
in file FreeCAD/src/Mod/TechDraw/App/LineGenerator.h
after
#include <QPen>
+#include <map>

and in file 
FreeCAD/build/src/Mod/CAM/PathSimulator/AppGL/CAMSimulator_autogen/EWIEGA46WW/../../../../../../../../src/Mod/CAM/PathSimulator/AppGL/DlgCAMSimulator.h

-#include <QtGui/qopenglpaintdevice.h>
+#include <QtOpenGL/qopenglpaintdevice.h>

And after that it building 

### Full version info

```shell
I use FreeCAD on master branch with Qt6
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Tarot67, 2024-05-23:
the text between '<' abd '>"=' hasbeen remove Grrr

After QPen
add
'#include <map>'

Tarot67, 2024-05-23:
map

Tarot67, 2024-05-23:
after QPen include add include map

maxwxyz, 2024-05-23:
@WandererFan can you confirm?

obelisk79, 2024-05-25:
confirmed issue, adding the missing include resolved the issue

closed 2024-06-10 by WandererFan