FreeCAD/FreeCAD #14247: spreadsheet used all ram

Vincenzoferrara, 2024-05-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

when I use the spreadsheet and use the "select all cells" button and then format them (align them to the center) it starts to use a lot of ram, and every time I modify a fly the amount of ram used increases.

I think the problem is that it also "sees" the empty cells as needing to be recompiled, this bug also exists in the stable version

(I use 32GB of RAM and an i7 12gen)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git) AppImage
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT) [ OS: Italian/Italy (it_IT) ]
Installed mods: 
  * AutoOpenSpreadsheet
  * ProDarkThemePreferencePack 1.0.0
  * MeshRemodel 1.9.23
  * fasteners.backup1715436808.37841 0.5.16 (Disabled)
  * MakerWorkbench 1.0.1
  * FeedsAndSpeeds 0.5.0
  * ExplodedAssembly
  * fasteners 0.5.21
  * Assembly4 0.50.12
  * CubeMenu
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-05-23:
A clarification: this is not about the Edit > Select All in a Spreadsheet, which apparently does nothing. This is bug is about clicking the upper left corner of a spreadsheet, which selects all 11075584 "possible" cells. Ideally there should be a way to select only "mapped" cells, as in only cells with content.

kaiwas, 2024-05-24:
I confirm.
This behavior occurs in all versions.
If select all and click "Delete" - the program eats up all the memory and freezes

xtemp09, 2024-05-24:
I'd say, this is seriously connected to #11335.

maxwxyz, 2024-05-24:
Yes

maxwxyz, 2024-05-24:
Duplicate of #11335

closed 2024-05-24 by maxwxyz