FreeCAD/FreeCAD #14251: modification of the length "Ext. Lines"

DBouquerel, 2024-05-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Draft -> Creation of a horizontal dimension -> modification of the length "Ext. Lines" -> Displacement of the dimension in height -> the change of the length "Ext. Lines" is no longer done. the length remains blocked on that previously established.

### Full version info

```shell
[code]
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-05-24:
How do you move the dimension line? This should be done with Draft_Edit, not with Draft_Move.

DBouquerel, 2024-05-24:
Ok, iI made the mistake of moving with “Draft_Move”. It seemed logical to me. Sorry.

Roy-043, 2024-05-25:
OK, thank for your feedback. I am closing this issue then.

closed 2024-05-25 by Roy-043