FreeCAD/FreeCAD #14253: BIM: buildingsmart-tech.org links in code no longer work

Roy-043, 2024-05-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Links in the code to buildingsmart-tech.org no longer work.
BimPreflight.py (1 link)
buildPsets.py (2 links)

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git)
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

3x380V, 2024-05-24:
A naive solution looks like:
```
diff --git a/src/Mod/BIM/bimcommands/BimPreflight.py b/src/Mod/BIM/bimcommands/BimPreflight.py
index 9e3434217c..4453f16fa6 100644
--- a/src/Mod/BIM/bimcommands/BimPreflight.py
+++ b/src/Mod/BIM/bimcommands/BimPreflight.py
@@ -894,7 +894,7 @@ class BIM_Preflight_TaskPanel:
                     "\n"
                     + translate(
                         "BIM",
-                        "Verify which properties a certain property set must contain on http://www.buildingsmart-tech.org/ifc/IFC4/Add2/html/annex/annex-b/alphabeticalorder_psets.htm",
+                        "Verify which properties a certain property set must contain on https://standards.buildingsmart.org/IFC/DEV/IFC4_2/FINAL/HTML/annex/annex-b/alphabeticalorder_psets.htm",
                     )
                     + "\n\n"
                 )
diff --git a/src/Mod/BIM/utils/buildPsets.py b/src/Mod/BIM/utils/buildPsets.py
index 4e8d508344..f00f41b9d5 100644
--- a/src/Mod/BIM/utils/buildPsets.py
+++ b/src/Mod/BIM/utils/buildPsets.py
@@ -33,7 +33,7 @@ MAXTRIES = 3
 # read the pset list
 print("Getting psets list...")
 u = urllib2.urlopen(
-    "http://www.buildingsmart-tech.org/ifc/IFC4/Add2/html/annex/annex-b/alphabeticalorder_psets.htm"
+    "https://standards.buildingsmart.org/IFC/DEV/IFC4_2/FINAL/HTML/annex/annex-b/alphabeticalorder_psets.htm"
 )
 p = u.read()
 u.close()
@@ -47,7 +47,7 @@ for i, pset in enumerate(psets):
     for j in range(MAXTRIES):
         try:
             u = urllib2.urlopen(
-                "http://www.buildingsmart-tech.org/ifc/IFC4/Add2/html/psd/Pset_"
+                "https://standards.buildingsmart.org/IFC/DEV/IFC4_2/FINAL/HTML/psd/Pset_"
                 + pset
                 + ".xml"
             )

```

yorikvanhavre, 2024-05-24:
The IFC documentation has the bad habit of changing locations quite often :)
And there is unfortunately apparently no "default" location that would "always work".

I think we could maybe define an IFC_DOCS_ROOT_URL variable somewhere to be used everywhere in BIM so it's easy to update next time

3x380V, 2024-05-24:
Agree. I can prepare PR in case you've not already done that.

yorikvanhavre, 2024-05-24:
That would be cool! Yes, please do! Thanks!!

3x380V, 2024-05-24:
Done as #14258. Please note that only `buildPsets.py` is tested while patch changing translation strings is not.

closed 2024-05-28 by yorikvanhavre