FreeCAD/FreeCAD #14254: FreeCAD crashes when loading an inappropriate project file

wwmayer, 2024-05-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In https://forum.freecad.org/viewtopic.php?t=87857 a problem is discussed with a missing camera. But when loading the project file [Kitchen fixed.FCStd](https://forum.freecad.org/download/file.php?id=264634) then FreeCAD crashes. 

While loading the file I got a few warnings and errors:
```
Cannot create object 'Import': ('Part::SubShapeBinder' is not a document object type)
Cannot create object 'Import001': ('Part::SubShapeBinder' is not a document object type)
Cannot create object 'Import002': ('Part::SubShapeBinder' is not a document object type)
Cannot create object 'Import003': ('Part::SubShapeBinder' is not a document object type)
<PropertyLinks> PropertyLinks.cpp(1018): Lost link to Kitchen_fixed Import while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(1018): Lost link to Kitchen_fixed Import001 while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(1018): Lost link to Kitchen_fixed Import002 while loading, maybe an object was not loaded correctly
Lost link to 'Import' while loading, maybe an object was not loaded correctly
Lost link to 'Import' while loading, maybe an object was not loaded correctly
Lost link to 'Import001' while loading, maybe an object was not loaded correctly
Lost link to 'Import001' while loading, maybe an object was not loaded correctly
Lost link to 'Import002' while loading, maybe an object was not loaded correctly
Lost link to 'Import002' while loading, maybe an object was not loaded correctly
<PropertyLinks> PropertyLinks.cpp(1018): Lost link to Kitchen_fixed Import003 while loading, maybe an object was not loaded correctly
Lost link to 'Import003' while loading, maybe an object was not loaded correctly
Lost link to 'Import003' while loading, maybe an object was not loaded correctly
Enumeration index -1 is out of range, ignore it
<-- Snip -->
Enumeration index -1 is out of range, ignore it
Coin read error: Couldn't read value for field "position" of OrthographicCamera
	Occurred at line   3 in <memory>
Camera settings failed to read
```

Then after a short time FreeCAD crashes with this callstack:
```
Program received signal SIGSEGV, Segmentation fault.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f08ecd49520]
#1  0x7f08d2675139 in boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u>::operator()() from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x39
#2  0x7f08d2674eb5 in unsigned long boost::random::detail::generate_uniform_int<boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u>, unsigned long>(boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u>&, unsigned long, unsigned long, mpl_::bool_<true>) from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x165
#3  0x7f08d2674d45 in unsigned long boost::random::detail::generate_uniform_int<boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u>, unsigned long>(boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u>&, unsigned long, unsigned long) from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x25
#4  0x7f08d2674d14 in unsigned long boost::random::uniform_int_distribution<unsigned long>::operator()<boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u> >(boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u>&) const from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x24
#5  0x7f08d2674cbd in unsigned long boost::uniform_int<unsigned long>::operator()<boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u> >(boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u>&) const from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x1d
#6  0x7f08d2674c1c in boost::random::variate_generator<boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u>*, boost::uniform_int<unsigned long> >::operator()() from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x2c
#7  0x7f08d266b2a7 in boost::uuids::basic_random_generator<boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u> >::operator()() from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x27
#8  0x7f08d2667741 in TechDraw::Vertex::createNewTag() from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x111
#9  0x7f08d2667a59 in TechDraw::Vertex::Vertex(double, double) from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x189
#10  0x7f08d2667ae2 in TechDraw::Vertex::Vertex(Base::Vector3<double>) from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x22
#11  0x7f08d267cd82 in void __gnu_cxx::new_allocator<TechDraw::Vertex>::construct<TechDraw::Vertex, Base::Vector3<double> >(TechDraw::Vertex*, Base::Vector3<double>&&) from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x62
#12  0x7f08d267cd0d in void std::allocator_traits<std::allocator<TechDraw::Vertex> >::construct<TechDraw::Vertex, Base::Vector3<double> >(std::allocator<TechDraw::Vertex>&, TechDraw::Vertex*, Base::Vector3<double>&&) from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x3d
#13  0x7f08d267cc98 in std::_Sp_counted_ptr_inplace<TechDraw::Vertex, std::allocator<TechDraw::Vertex>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<Base::Vector3<double> >(std::allocator<TechDraw::Vertex>, Base::Vector3<double>&&) from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x88
#14  0x7f08d267cb58 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<TechDraw::Vertex, std::allocator<TechDraw::Vertex>, Base::Vector3<double> >(TechDraw::Vertex*&, std::_Sp_alloc_shared_tag<std::allocator<TechDraw::Vertex> >, Base::Vector3<double>&&) from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x98
#15  0x7f08d267caa0 in std::__shared_ptr<TechDraw::Vertex, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<TechDraw::Vertex>, Base::Vector3<double> >(std::_Sp_alloc_shared_tag<std::allocator<TechDraw::Vertex> >, Base::Vector3<double>&&) from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x50
#16  0x7f08d267ca3d in std::shared_ptr<TechDraw::Vertex>::shared_ptr<std::allocator<TechDraw::Vertex>, Base::Vector3<double> >(std::_Sp_alloc_shared_tag<std::allocator<TechDraw::Vertex> >, Base::Vector3<double>&&) from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x3d
#17  0x7f08d267c9f0 in std::shared_ptr<TechDraw::Vertex> std::allocate_shared<TechDraw::Vertex, std::allocator<TechDraw::Vertex>, Base::Vector3<double> >(std::allocator<TechDraw::Vertex> const&, Base::Vector3<double>&&) from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x40
#18  0x7f08d267c091 in std::shared_ptr<TechDraw::Vertex> std::make_shared<TechDraw::Vertex, Base::Vector3<double> >(Base::Vector3<double>&&) from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x41
#19  0x7f08d2679e06 in TechDraw::GeometryObject::addGeomFromCompound(TopoDS_Shape, TechDraw::edgeClass, bool) from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x5c6
#20  0x7f08d2677ff6 in TechDraw::GeometryObject::extractGeometry(TechDraw::edgeClass, bool) from /home/werner/Projects/build_clang/Mod/TechDraw/TechDraw.so+0x256
[... cut ...]
### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37462 (Git)
Build type: debug
Branch: main
Hash: d12635246f86947031776a42714028b236bd05ed
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

None

### Anything else?

It must be check if the issue with the lost camera is related to the crash.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-05-24:
I cannot reproduce the crash. Perhaps, it depends on some options in _TechDraw_.

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37303 (Git)
Build type: Debug
Branch: main
Hash: 869cb1f83b5f92ee134dafde46c9c5e0dceb8889
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

maxwxyz, 2024-05-24:
@bgbsww could it be related to https://github.com/FreeCAD/FreeCAD/pull/14202

bgbsww, 2024-05-24:
I don't think so.  I'll note, however, that Part::SubShapeBinder exists in LS3, but does not in main ( implementation is fully in PartDesign, not Part ).  So I suspect this file was edited in LS3 with subshapebinders added, saved, and then attempted to open in main to this failure.

That's a semi-educated guess, could be wrong.

wwmayer, 2024-05-24:
>  I'll note, however, that Part::SubShapeBinder exists in LS3

That's correct. In the meantime it is confirmed in the forum thread that the file was created with RT's version. Nevertheless, the application must not crash and according to the callstack I get the crash is not directly related to the missing SubShapeBinder.

xtemp09, 2024-05-24:
If I open a sketch, _FreeCAD_ crashes, the backtrace is different.

<details><summary>Backtrace</summary>

```cpp
Process 22374 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x88)
    frame #0: 0x00007ffff60a47f5 libFreeCADApp.so`Data::ElementMap::dehashElementName(this=0x0000000000000000, name=0x00007fffffff8260) const at ElementMap.cpp:708:5
   705      if (name.empty()) {
   706          return name;
   707      }
-> 708      if (!this->hasher) {
   709          return name;
   710      }
   711      auto id = App::StringID::fromString(name.toRawBytes());
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x88)
  * frame #0: 0x00007ffff60a47f5 libFreeCADApp.so`Data::ElementMap::dehashElementName(this=0x0000000000000000, name=0x00007fffffff8260) const at ElementMap.cpp:708:5
    frame #1: 0x00007ffff60a78fd libFreeCADApp.so`Data::ElementMap::getElementHistory(this=0x0000000000000000, name=0x000055555e12eb78, masterTag=5019, original=<unavailable>, history=size=0) const at ElementMap.cpp:1338:36
    frame #2: 0x00007fffbf13b2aa Part.so`::getElementSource(App::DocumentObject *, Part::TopoShape, const Data::MappedName &, char) [inlined] Data::ComplexGeoData::getElementHistory(history=size=0, original=0x00007fffffff84e0, name=<unavailable>, this=0x00007fffffff8a00) const at ComplexGeoData.h:299:46
    frame #3: 0x00007fffbf13b28f Part.so`getElementSource(owner=<unavailable>, shape=TopoShape @ 0x00007fffffff8a00, name=0x00007fffffff8820, type='F') at PartFeature.cpp:467:43
    frame #4: 0x00007fffbf13b514 Part.so`Part::Feature::getRelatedElements(obj=<unavailable>, name=";#c23;:G0;XTR;:H139a:8,F.Face12", sameType=followTypeChange, withCache=false) at PartFeature.cpp:778:91
    frame #5: 0x00007fffbf2cc008 Part.so`Attacher::AttachEngine::calculateAttachedPlacement(this=0x000055555bb96660, origPlacement=0x000055555bb99660, subChanged=0x0000000000000000) at Attacher.cpp:971:63
    frame #6: 0x00007fffbf154b36 Part.so`Part::AttachExtension::isAttacherActive(this=0x000055555bb99968) const at AttachExtension.cpp:312:56
    frame #7: 0x00007fffbf157113 Part.so`Part::AttachExtension::extensionOnChanged(this=0x000055555bb99968, prop=0x000055555bb99b08) at AttachExtension.cpp:366:50
    frame #8: 0x00007ffff5eeea12 libFreeCADApp.so`App::ExtensionContainer::onChanged(this=0x000055555bb99130, prop=0x000055555bb99b08) at ExtensionContainer.cpp:282:41
    frame #9: 0x00007ffff5ee9bf7 libFreeCADApp.so`App::DocumentObject::onChanged(this=0x000055555bb99130, prop=0x000055555bb99b08) at DocumentObject.cpp:818:35
    frame #10: 0x00007ffff5f906fc libFreeCADApp.so`App::GeoFeature::onChanged(this=0x000055555bb99130, prop=0x000055555bb99b08) at GeoFeature.cpp:255:30
    frame #11: 0x00007fffbf13c13c Part.so`Part::Feature::onChanged(this=0x000055555bb99130, prop=0x000055555bb99b08) at PartFeature.cpp:1469:26
    frame #12: 0x00007fffba34130e _PartDesign.so`PartDesign::Plane::onChanged(this=0x000055555bb99130, prop=0x000055555bb99b08) at DatumPlane.cpp:101:21
    frame #13: 0x00007ffff5fe12e2 libFreeCADApp.so`App::Property::hasSetValue(this=0x000055555bb99b08) at Property.cpp:230:26
    frame #14: 0x00007ffff60161a6 libFreeCADApp.so`App::PropertyLinkBase::hasSetValue(this=0x000055555bb99b08) at PropertyLinks.cpp:78:26
    frame #15: 0x00007ffff600a5bf libFreeCADApp.so`App::PropertyLinkSubList::setValues(this=0x000055555bb99b08, lValue=size=0, lSubNames=size=0, ShadowSubList=size=0) at PropertyLinks.cpp:2047:16
    frame #16: 0x00007ffff600a658 libFreeCADApp.so`App::PropertyLinkSubList::setValues(this=0x000055555bb99b08, lValue=size=1, lSubNames=size=1, ShadowSubList=size=0) at PropertyLinks.cpp:2001:14
    frame #17: 0x00007ffff600aa85 libFreeCADApp.so`App::PropertyLinkSubList::Paste(this=0x000055555bb99b08, from=0x000055555bb999f8) at PropertyLinks.cpp:2725:14
    frame #18: 0x00007fffbf1570dd Part.so`Part::AttachExtension::extensionOnChanged(this=0x000055555bb99968, prop=0x000055555bb999f8) at AttachExtension.cpp:354:40
    frame #19: 0x00007ffff5eeea12 libFreeCADApp.so`App::ExtensionContainer::onChanged(this=0x000055555bb99130, prop=0x000055555bb999f8) at ExtensionContainer.cpp:282:41
    frame #20: 0x00007ffff5ee9bf7 libFreeCADApp.so`App::DocumentObject::onChanged(this=0x000055555bb99130, prop=0x000055555bb999f8) at DocumentObject.cpp:818:35
    frame #21: 0x00007ffff5f906fc libFreeCADApp.so`App::GeoFeature::onChanged(this=0x000055555bb99130, prop=0x000055555bb999f8) at GeoFeature.cpp:255:30
    frame #22: 0x00007fffbf13c13c Part.so`Part::Feature::onChanged(this=0x000055555bb99130, prop=0x000055555bb999f8) at PartFeature.cpp:1469:26
    frame #23: 0x00007fffba34130e _PartDesign.so`PartDesign::Plane::onChanged(this=0x000055555bb99130, prop=0x000055555bb999f8) at DatumPlane.cpp:101:21
    frame #24: 0x00007ffff5fe12e2 libFreeCADApp.so`App::Property::hasSetValue(this=0x000055555bb999f8) at Property.cpp:230:26
    frame #25: 0x00007ffff60161a6 libFreeCADApp.so`App::PropertyLinkBase::hasSetValue(this=0x000055555bb999f8) at PropertyLinks.cpp:78:26
    frame #26: 0x00007ffff60118c8 libFreeCADApp.so`App::PropertyLinkSubList::updateElementReference(this=0x000055555bb999f8, feature=0x000055555bb7c620, reverse=false, notify=true) at PropertyLinks.cpp:2418:20
[... cut ...]
    frame #129: 0x00007ffff3229e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:105:1), argc=1, argv=0x00007fffffffdd98, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd88) at libc-start.c:392:3
    frame #130: 0x0000555555569c05 FreeCAD`_start + 37
(lldb)
```

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37464 (Git)
Build type: Debug
Branch: main
Hash: db48a78d5a545bb21a5c80a9fa4b4c983fbbe951
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

bgbsww, 2024-05-24:
Okay, so narrowly defined, no, this isn't related to the unmerged #14202.  That second stack trace hits buildExtrusion which is new, and also PropertyLinkSubList, and either of those might be TNP related.  Not questioning the importance of not crashing, just not sure where the issues lie.

bgbsww, 2024-06-04:
Can we get a double check on if this still happens?  I can no longer reproduce, and either I or someone else may have inadvertently fixed this without tagging it in our PR.

wwmayer, 2024-06-04:
I have tested it with the latest version and didn't get a crash any more. After rebuilding d12635246f I still couldn't get it to crash any more.

However, when looking at the callstack you will find `boost::uuids::basic_random_generator` and according to the [SO](https://stackoverflow.com/questions/18890355/is-boostuuidsrandom-generator-thread-safe) article the function is not thread-safe.
And you can easily see that these functions are called from within a worker thread and not the main thread.

luzpaz, 2024-10-12:
Bumping for visibility