FreeCAD/FreeCAD #14255: Sketch not visible on body surface.

Panda631, 2024-05-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a sketch on an existing surface of a body, it is not visible with standart color settings.
The rendered top color of the body seems to be identical to the sketch color. 

[image]

In the example above, there is a sketch with a rectangle.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git)
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly3 0.12.0
  * Assembly4 0.50.4
  * EM
  * fasteners 0.4.70
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons
  * Render 2023.8.18
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-05-24:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/13989

maxwxyz, 2024-05-24:
@Panda631 please use the search function before opening an issue, you've checked the box...

closed 2024-05-24 by maxwxyz