FreeCAD/FreeCAD #14260: Body rotation in draft workbench - no firstangle

IronFist16, 2024-05-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello,

While trying to assemble step files exported from separate design (FCStd) files, intitally I could do body movement and rotation but all the sudden I started receiving the below error message and I cannot visually see the movement/rotation, until I type a figure (distance, angle), which does not allow me to move/rotate as per the snap constraints,

Can anyone please let me know how to fix this issue?

```
20:05:55  GuiCommand: Rotate
20:05:55  Pick rotation center
20:05:56  Pick base angle
20:05:57  Pick rotation angle
20:05:57  <class 'AttributeError'>
20:05:57  Traceback (most recent call last):
20:05:57    File "/usr/share/freecad/Mod/Draft/draftguitools/gui_rotate.py", line 115, in action
20:05:57      self.handle_mouse_move_event(arg)
20:05:57    File "/usr/share/freecad/Mod/Draft/draftguitools/gui_rotate.py", line 160, in handle_mouse_move_event
20:05:57      if angle < self.firstangle:
20:05:58  <class 'AttributeError'>
20:05:58  Traceback (most recent call last):
20:05:58    File "/usr/share/freecad/Mod/Draft/draftguitools/gui_rotate.py", line 119, in action
20:05:58      self.handle_mouse_click_event(arg)
20:05:58    File "/usr/share/freecad/Mod/Draft/draftguitools/gui_rotate.py", line 182, in handle_mouse_click_event
20:05:58      self.set_rotation_angle(arg)
20:05:58    File "/usr/share/freecad/Mod/Draft/draftguitools/gui_rotate.py", line 224, in set_rotation_angle
20:05:58      if angle < self.firstangle:
20:05:58  AttributeError: 'Rotate' object has no attribute 'firstangle'
```

### Full version info

```shell
[code]
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Branch: unknown
Hash: 1397818c9664d158281339a84be59d3e9887f6ad
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.12
  * Defeaturing 1.2.0
  * A2plus 0.4.64a
  * freecad.gears 1.0.0
[/code]
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-24:
Could you please try this with the last stable or dev version, you are using an outdated version, maybe the files were created with a newer version.

IronFist16, 2024-05-25:
I've tried with recent version, same issue.
On my previous attempt, the Steps were exported from same version.

maxwxyz, 2024-05-25:
@Roy-043 FYI

Roy-043, 2024-05-25:
"All of a sudden" scenarios are difficult for obvious reasons. @IronFist16 can you post an example file and also step-by-step instruction on how to reproduce the issue? Thanks.

Roy-043, 2024-06-06:
I am closing this one. @IronFist16 feel free to reopen if you have additional info.

closed 2024-06-06 by Roy-043