FreeCAD/FreeCAD #14261: Unable to install shapefile.py library on MacOS

tmcsys, 2024-05-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Python NOOB here.

Trying to install shapefile.py library from the Import dialog.

```
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/Applications/FreeCAD.app/Contents/Resources/Mod/Arch/importSHP.py", line 49, in insert
    if not checkShapeFileLibrary():
File "/Applications/FreeCAD.app/Contents/Resources/Mod/Arch/importSHP.py", line 139, in checkShapeFileLibrary
    u = addonmanager_utilities.urlopen(url)
<class 'AttributeError'>: module 'addonmanager_utilities' has no attribute ‘urlopen'
```

### Full version info

```shell
[code]
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-25:
Help requests please in on forum. You can try it if it works in the latest weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

tmcsys, 2024-05-29:
Could we reopen this please?  While a Python NOOB, I am an experienced developer.

Or maybe I should create a new issue specifically identifying the defect.

I tried the latest dev version with exactly the same results:
```
15:08:02  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Volumes/FreeCAD_weekly-builds-37543-conda-macOS-x86_64-py311/FreeCAD.app/Contents/Resources/Mod/BIM/importers/importSHP.py", line 49, in insert
    if not checkShapeFileLibrary():
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/Volumes/FreeCAD_weekly-builds-37543-conda-macOS-x86_64-py311/FreeCAD.app/Contents/Resources/Mod/BIM/importers/importSHP.py", line 139, in checkShapeFileLibrary
    u = addonmanager_utilities.urlopen(url)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: module 'addonmanager_utilities' has no attribute 'urlopen'
```
The problem is exactly what is stated in the error message.  **There is no urlopen routine in Mod/AddonManager/addonmanager_utilities.**

I suspect the problem is in Mod/BIM/importers/importSHP.py. If you search the Mod/BIM/importers, importSHP.py is the only file that references addonmanager_utilities.urlopen. 

```
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

maxwxyz, 2024-05-30:
@yorikvanhavre FYI

yorikvanhavre, 2024-05-30:
oops, addon manager has apparently changed :sweat_smile:
I'll submit a fix

PR #14385 (2024-05-30): BIM: Fixed import in importSHP
Changed obsolete import - Fixes #14261
commit f11b08bd (2024-05-30): BIM: Fixed import in importSHP - fixes #14261 (#14385)
closed 2024-05-30 by Roy-043