FreeCAD/FreeCAD #14262: BIM: error while adding stairs

balrobs, 2024-05-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

adding a stair I see the following error:
```
18:42:15  Vector (0.0, 0.0, 0.0) Vector (4500.0, 0.0, 3000.0)
18:42:15  Running the Python command 'Arch_Stairs' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD_weekly-builds-37358-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimStairs.py", line 65, in Activated
    FreeCADGui.doCommand("obj = Arch.makeStairs(steps="+str(params.get_param_arch("StairsSteps"))+")")
  File "<string>", line 1, in <module>
  File "C:\Program Files\FreeCAD_weekly-builds-37358-conda-Windows-x86_64-py311\Mod\BIM\Arch.py", line 850, in makeStairs
    makeRailing(stairs)
  File "C:\Program Files\FreeCAD_weekly-builds-37358-conda-Windows-x86_64-py311\Mod\BIM\Arch.py", line 901, in makeRailing
    makeRailingLorR(stairs,"L")
  File "C:\Program Files\FreeCAD_weekly-builds-37358-conda-Windows-x86_64-py311\Mod\BIM\Arch.py", line 875, in makeRailingLorR
    lrRail = Arch.makePipe(baseobj=None,diameter=0,length=0,placement=None,name=translate("Arch","Railing"))
             ^^^^
name 'Arch' is not defined
```

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git)
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Reinforcement
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-25:
@yorikvanhavre FYI

paullee0, 2024-05-25:
Confirmed, same here on :
- FreeCAD_weekly-builds-37436-conda-Linux-x86_64-py311.AppImage
- on Fedora 38

balrobs, 2024-05-25:
Hi @paullee0, can confirm your fix works. Thank you very much! 👍

Roy-043, 2024-05-25:
Completed with #14271.

closed 2024-05-25 by Roy-043