FreeCAD/FreeCAD #14268: FreeCAD crashes with SIGSEGV when opening a specific file

ttomasini, 2024-05-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Opening [testfile.FCStd](https://f3l.de/bugreports/bird_home_holder.FCStd) in the current dev-0.22.0-37464 version of freecad causes a crash of the application with the following macOS error. I stripped the error a bit to be able to focus on the important parts. I hope that I did not miss anything. Please let me know if you need more information. The crash is reproducible and any other file opens just fine.

```
Process:               freecad [52881]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        Exited process [52875]

Date/Time:             2024-05-25 09:30:19.8737 +0200
OS Version:            macOS 13.6.6 (22G630)
Report Version:        12
Bridge OS Version:     8.4 (21P4222)

Time Awake Since Boot: 130000 seconds
Time Since Wake:       1299 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000058
Exception Codes:       0x0000000000000001, 0x0000000000000058

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [52881]

VM Region Info: 0x58 is not in any region.  Bytes before following region: 140737488138152
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      shared memory            7ffffffcb000-7ffffffcc000 [    4K] r-x/r-x SM=SHM  

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libFreeCADApp.dylib           	       0x106e4623e Data::ElementMap::dehashElementName(Data::MappedName const&) const + 62
1   libFreeCADApp.dylib           	       0x106e4c27c Data::ElementMap::getElementHistory(Data::MappedName const&, long, Data::MappedName*, std::__1::vector<Data::MappedName, std::__1::allocator<Data::MappedName>>*) const + 1116
2   Part.so                       	       0x11721055c getElementSource(App::DocumentObject*, Part::TopoShape, Data::MappedName const&, char) + 444
3   Part.so                       	       0x11720fad5 Part::Feature::getRelatedElements(App::DocumentObject*, char const*, Part::HistoryTraceType, bool) + 341
4   Part.so                       	       0x1173d5009 Attacher::AttachEngine::calculateAttachedPlacement(Base::Placement const&, bool*) + 377
5   Part.so                       	       0x1172357dc Part::AttachExtension::isAttacherActive() const + 108
6   Part.so                       	       0x1172359da Part::AttachExtension::extensionOnChanged(App::Property const*) + 250
7   libFreeCADApp.dylib           	       0x106c2edd8 App::ExtensionContainer::onChanged(App::Property const*) + 72
8   libFreeCADApp.dylib           	       0x106c2637d App::DocumentObject::onChanged(App::Property const*) + 1293
9   Part.so                       	       0x11721566c Part::Feature::onChanged(App::Property const*) + 1036
10  Sketcher.so                   	       0x11c232b2a Sketcher::SketchObject::onChanged(App::Property const*) + 378
11  libFreeCADApp.dylib           	       0x106d3c45b App::Property::hasSetValue() + 59
12  libFreeCADApp.dylib           	       0x106d716c9 App::PropertyLinkSubList::setValues(std::__1::vector<App::DocumentObject*, std::__1::allocator<App::DocumentObject*>>&&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>&&, std::__1::vector<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>>&&) + 937
13  libFreeCADApp.dylib           	       0x106d71254 App::PropertyLinkSubList::setValues(std::__1::vector<App::DocumentObject*, std::__1::allocator<App::DocumentObject*>> const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const&, std::__1::vector<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>>&&) + 164
14  libFreeCADApp.dylib           	       0x106d79ebc App::PropertyLinkSubList::Paste(App::Property const&) + 172
15  Part.so                       	       0x117235c4d Part::AttachExtension::onExtendedDocumentRestored() + 45
16  libFreeCADApp.dylib           	       0x106c281cc App::DocumentObject::onDocumentRestored() + 76
17  Sketcher.so                   	       0x11c232f07 Sketcher::SketchObject::onDocumentRestored() + 23
18  libFreeCADApp.dylib           	       0x106b91f37 App::Document::afterRestore(std::__1::vector<App::DocumentObject*, std::__1::allocator<App::DocumentObject*>> const&, bool) + 2391
19  libFreeCADApp.dylib           	       0x106b961f3 App::Document::afterRestore(bool) + 83
20  libFreeCADApp.dylib           	       0x106ddee0c App::Application::openDocuments(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>*, bool) + 8204
21  libFreeCADApp.dylib           	       0x106ddcd3a App::Application::openDocument(char const*, bool) + 218
22  libFreeCADApp.dylib           	       0x106e14b46 App::Application::sOpenDocument(_object*, _object*, _object*) + 294
23  libpython3.11.dylib           	       0x1047ba4c8 cfunction_call + 56
24  libpython3.11.dylib           	       0x1048b02c6 _PyEval_EvalFrameDefault + 235078
25  libpython3.11.dylib           	       0x104874ceb _PyEval_Vector + 3003
26  libpython3.11.dylib           	       0x1048740b9 PyEval_EvalCode + 249
27  libpython3.11.dylib           	       0x104922fc8 run_mod + 168
28  libpython3.11.dylib           	       0x1049252f8 PyRun_StringFlags + 104
29  libFreeCADBase.dylib          	       0x1043c7bff Base::InterpreterSingleton::runString(char const*) + 111
30  libFreeCADApp.dylib           	       0x106e14730 App::Application::sLoadFile(_object*, _object*) + 1712
31  libpython3.11.dylib           	       0x1047ba4ee cfunction_call + 94
32  libpython3.11.dylib           	       0x1048b02c6 _PyEval_EvalFrameDefault + 235078
33  libpython3.11.dylib           	       0x104874ceb _PyEval_Vector + 3003
34  libpython3.11.dylib           	       0x1048740b9 PyEval_EvalCode + 249
35  libpython3.11.dylib           	       0x104922fc8 run_mod + 168
36  libpython3.11.dylib           	       0x1049252f8 PyRun_StringFlags + 104
37  libFreeCADBase.dylib          	       0x1043c7bff Base::InterpreterSingleton::runString(char const*) + 111
38  StartGui.so                   	       0x11426e5a5 StartGui::StartView::fileCardSelected(QModelIndex const&) + 549
39  libQt5Core.5.15.13.dylib      	       0x107c58d37 0x107b0c000 + 1363255
40  libQt5Widgets.5.15.13.dylib   	       0x107338fcb QAbstractItemView::clicked(QModelIndex const&) + 45
```

### Full version info

```shell
OS: macOS 13.6.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37464 (Git)
Build type: Release
Branch: main
Hash: db48a78d5a545bb21a5c80a9fa4b4c983fbbe951
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * Manipulator 1.5.7
  * fasteners 0.5.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-25:
related? https://github.com/FreeCAD/FreeCAD/issues/14254

xtemp09, 2024-05-25:
I've opened the test file and got the same backtrace. The backtrace is seriously connected to the one provided by me in [here](https://github.com/FreeCAD/FreeCAD/issues/14254#issuecomment-2129852507).

<details><summary>The backtrace</summary>

```cpp
Process 1702 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x88)
    frame #0: 0x00007ffff60a47f5 libFreeCADApp.so`Data::ElementMap::dehashElementName(this=0x0000000000000000, name=0x00007fffffff9810) const at ElementMap.cpp:708:5
   705      if (name.empty()) {
   706          return name;
   707      }
-> 708      if (!this->hasher) {
   709          return name;
   710      }
   711      auto id = App::StringID::fromString(name.toRawBytes());
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x88)
  * frame #0: 0x00007ffff60a47f5 libFreeCADApp.so`Data::ElementMap::dehashElementName(this=0x0000000000000000, name=0x00007fffffff9810) const at ElementMap.cpp:708:5
    frame #1: 0x00007ffff60a78fd libFreeCADApp.so`Data::ElementMap::getElementHistory(this=0x0000000000000000, name=0x000055555a604a78, masterTag=742, original=<unavailable>, history=size=0) const at ElementMap.cpp:1338:36
    frame #2: 0x00007fffc313b2aa Part.so`::getElementSource(App::DocumentObject *, Part::TopoShape, const Data::MappedName &, char) [inlined] Data::ComplexGeoData::getElementHistory(history=size=0, original=0x00007fffffff9a90, name=<unavailable>, this=0x00007fffffff9fb0) const at ComplexGeoData.h:299:46
    frame #3: 0x00007fffc313b28f Part.so`getElementSource(owner=<unavailable>, shape=TopoShape @ 0x00007fffffff9fb0, name=0x00007fffffff9dd0, type='F') at PartFeature.cpp:467:43
    frame #4: 0x00007fffc313b514 Part.so`Part::Feature::getRelatedElements(obj=<unavailable>, name=";g15;SKT;:H2da,E;:G;XTR;:H2da:7,F;:H-2dc,F;:H2dc,F;:H2d7,F.Face9", sameType=followTypeChange, withCache=false) at PartFeature.cpp:778:91
    frame #5: 0x00007fffc32cc008 Part.so`Attacher::AttachEngine::calculateAttachedPlacement(this=0x000055555a5a6b70, origPlacement=0x000055555a5a9ed0, subChanged=0x0000000000000000) at Attacher.cpp:971:63
    frame #6: 0x00007fffc3154b36 Part.so`Part::AttachExtension::isAttacherActive(this=0x000055555a5aa1d8) const at AttachExtension.cpp:312:56
    frame #7: 0x00007fffc3157113 Part.so`Part::AttachExtension::extensionOnChanged(this=0x000055555a5aa1d8, prop=0x000055555a5aa378) at AttachExtension.cpp:366:50
    frame #8: 0x00007ffff5eeea12 libFreeCADApp.so`App::ExtensionContainer::onChanged(this=0x000055555a5a99a0, prop=0x000055555a5aa378) at ExtensionContainer.cpp:282:41
    frame #9: 0x00007ffff5ee9bf7 libFreeCADApp.so`App::DocumentObject::onChanged(this=0x000055555a5a99a0, prop=0x000055555a5aa378) at DocumentObject.cpp:818:35
    frame #10: 0x00007ffff5f906fc libFreeCADApp.so`App::GeoFeature::onChanged(this=0x000055555a5a99a0, prop=0x000055555a5aa378) at GeoFeature.cpp:255:30
    frame #11: 0x00007fffc313c13c Part.so`Part::Feature::onChanged(this=0x000055555a5a99a0, prop=0x000055555a5aa378) at PartFeature.cpp:1469:26
    frame #12: 0x00007fffb6fef832 Sketcher.so`Sketcher::SketchObject::onChanged(this=0x000055555a5a99a0, prop=0x000055555a5aa378) at SketchObject.cpp:8938:34
    frame #13: 0x00007ffff5fe12e2 libFreeCADApp.so`App::Property::hasSetValue(this=0x000055555a5aa378) at Property.cpp:230:26
    frame #14: 0x00007ffff60161a6 libFreeCADApp.so`App::PropertyLinkBase::hasSetValue(this=0x000055555a5aa378) at PropertyLinks.cpp:78:26
    frame #15: 0x00007ffff600a5bf libFreeCADApp.so`App::PropertyLinkSubList::setValues(this=0x000055555a5aa378, lValue=size=0, lSubNames=size=0, ShadowSubList=size=0) at PropertyLinks.cpp:2047:16
    frame #16: 0x00007ffff600a658 libFreeCADApp.so`App::PropertyLinkSubList::setValues(this=0x000055555a5aa378, lValue=size=1, lSubNames=size=1, ShadowSubList=size=0) at PropertyLinks.cpp:2001:14
    frame #17: 0x00007ffff600aa85 libFreeCADApp.so`App::PropertyLinkSubList::Paste(this=0x000055555a5aa378, from=0x000055555a5aa268) at PropertyLinks.cpp:2725:14
    frame #18: 0x00007fffc3157250 Part.so`Part::AttachExtension::onExtendedDocumentRestored(this=0x000055555a5aa1d8) at AttachExtension.cpp:399:36
    frame #19: 0x00007ffff5ee7279 libFreeCADApp.so`App::DocumentObject::onDocumentRestored(this=0x000055555a5a99a0) at DocumentObject.cpp:1121:40
    frame #20: 0x00007fffb6fd71a0 Sketcher.so`Sketcher::SketchObject::onDocumentRestored(this=0x000055555a5a99a0) at SketchObject.cpp:9050:47
    frame #21: 0x00007ffff5e7dea1 libFreeCADApp.so`App::Document::afterRestore(this=0x0000555558d0f340, objArray=<unavailable>, checkPartial=<unavailable>) at Document.cpp:2172:36
    frame #22: 0x00007ffff5e7f3a1 libFreeCADApp.so`App::Document::afterRestore(this=0x0000555558d0f340, checkPartial=<unavailable>) at Document.cpp:2116:21
    frame #23: 0x00007ffff606578b libFreeCADApp.so`App::Application::openDocuments(this=0x0000555555718530, filenames=size=1, paths=size=0, labels=size=0, errs=size=0, createView=<unavailable>) at Application.cpp:885:33
    frame #24: 0x00007ffff6066275 libFreeCADApp.so`App::Application::openDocument(this=0x0000555555718530, FileName=<unavailable>, createView=<unavailable>) at Application.cpp:684:79
    frame #25: 0x00007ffff6083164 libFreeCADApp.so`App::Application::sOpenDocument((null)=<unavailable>, args=<unavailable>, kwd=<unavailable>) at ApplicationPy.cpp:263:46
[... cut ...]
    frame #86: 0x00007ffff3229e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:105:1), argc=1, argv=0x00007fffffffdd98, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd88) at libc-start.c:392:3
    frame #87: 0x0000555555569c05 FreeCAD`_start + 37
(lldb)
```

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37464 (Git)
Build type: Debug
Branch: main
Hash: db48a78d5a545bb21a5c80a9fa4b4c983fbbe951
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

maxwxyz, 2024-05-25:
so a duplicate?

xtemp09, 2024-05-25:
The issue #14254 was opened with the backtrace different from the one provided in here.

wwmayer, 2024-05-26:
The crash is caused inside getElementHistory() (defined in ComplexGeoData.h) because the member variable `_elementMap` is a null pointer.
@bgbsww 
How should this be handled? Should it throw an exception or should it return a value (e.g. -1, 0 or whatever) to indicate an error?

wwmayer, 2024-05-26:
> so a duplicate?

There is no relation as far as I can see.

bgbsww, 2024-06-03:
In general, _elementMap was refactored out into a separate object, and its original initialization that occurred in SetElementName now sometimes has to be done elsewhere; there is an idiomatic code snippet:
```
                // Originally in ComplexGeoData::setElementName
                // LinkStable/src/App/ComplexGeoData.cpp#L1631
                // No longer possible after map separated in ElementMap.cpp

                if (!elementMap()) {
                    resetElementMap(std::make_shared<Data::ElementMap>());
                }

```
So in the original TNP code, this case is impossible because there is no pointer to be null.
This seems to be a case where history is being called on a shape that never had the _elementMap initialized.  We just force the _elementMap to be initialized and empty everywhere else, but it would be useful to understand how this happens in case there's another issue related to it.  I'm going to go look at that now ...

PR #14489 (2024-06-04): Toponaming: defend against uninitialized elementMap in getElementHistory
Fix #14268.
closed 2024-06-04 by chennes