FreeCAD/FreeCAD #14269: FreeCad - Mesure angle problem

Pendelf, 2024-05-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When measuring the angle, the program shows the wrong value

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37464 (Git)
Build type: Release
Branch: main
Hash: db48a78d5a545bb21a5c80a9fa4b4c983fbbe951
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.21
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * PieMenu 1.5.0
  * Pyramids-and-Polyhedrons
  * sheetmetal.backup1704622370.7006474 0.3.12
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

Measure
https://youtu.be/5TGJqrpUZ6A
[Angel_problem.zip](https://github.com/FreeCAD/FreeCAD/files/15442443/Angel_problem.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-25:
I guess it currently shows the smallest angle between the two lines which is indeed 30°
@hlorus can you confirm?

hlorus, 2024-05-25:
MeasureAngle was ported from part's DimensionAngular so it would be good to compare it to the behavior of the old measurement. IIRC it measures the angle between the normals of the elements which can be a bit confusing.
Some lines which represent the normals could help with that.

MiqCG, 2024-10-29:
> it would be good to compare it to the behavior of the old measurement.

Checked the provided example file in 0.21.1 and the Part Design "Measure Angular" tool and it is displayed with the correct angle (not the complementary one) between the selected segments and with the arc arrow that clarifies what has been measured. See pictures.

BTW, very nice guitar intro in the vídeo, Pendelf.

[image]

[image]

`[code]
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * fasteners 0.5.29
[/code]
`

karliss, 2025-03-18:
Testing with 1.1dev and the behavior seems inconsistent. For a prism it more or less always chose complementary angle, but the visualization is placed incorrectly. If it chooses the complementary angle it should at least draw the visualization appropriately. 
[image]

When measuring a polyhedron generated using "Pyramids and polyhedrons" addon result seems to be somewhat random but repeatable (probably based on the vertex order in the underlying data structures). Sometimes it picks correct angle, sometimes complementary one.
[image]

When measuring angle between 2 edges that have common vertex I would expect it to always pick the side where angle is less than 180° and never pick complementary angle. If the 2 edges don't share a vertex  choice between 4 quadrants is less  clear, but it should still visualize it in a way that's consistent with chosen angle.

edwilliams16, 2025-04-18:
Here's a simple example where the complementary angle is displayed.

[image]
It's correct in the underlying sketch:

[image]

[badanglemeasure.zip](https://github.com/user-attachments/files/19815438/badanglemeasure.zip)

1.1.0dev-41264 Mac Arm

luzpaz, 2026-01-20:
@nishendra3 any chance you could look into this ?

closed 2026-02-01 by maxwxyz