FreeCAD/FreeCAD #14274: [Feature request][Part Design] There is no additive equivalent tool for Hole (Stud).

mno13, 2024-05-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As in title, there is no additive equivalent tool for Hole. It could be called "Stud".  As counter intuitive it seems on first look, it could be very useful. Two examples:

- design of plastic components
- design of complicated parts, using boolean operation - this way holes could be created in "tool body".

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git)
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * boltsfc 2022.11.5
  * Freecad-Built-in-themes-beta 1.2.2
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-25:
what exactly do you mean by stud? An outer thread?

mno13, 2024-05-26:
No, thread should have separate "dress-up" tool that works on internal and external cylindrical surfaces. It should add cosmetic (or even have an option for physical representation for 3d printing) thread that could be referenced in drawing. As Freecad doesn't have any thread representation yet, it's separate problem.

I have in mind exact additive equivalent to hole feature.  Name could be "Stud" (as threaded rod) or "Boss". "Boss" feature is popular in plastic design, but this tool in other systems is more complicated.

Boss feature example - there often are also tapered walls, and additional holes. Inverse hole could be first step for that:
[image]

Below is a practical example for "inverse hole" usage in design. Hydraulic manifold with lot of drilled oil channels. One of design practice is to design all holes in separate ("positive") body, and then boolean cut it from base shape (often raw casting).

[image]

FEA-eng, 2024-05-26:
Is it something you've seen in other CAD software? IMO this related feature would be more needed: #13536

Your proposal would be to make a tool that would still use a sketched circle like the Hole tool and just allowed to change its diameter without editing the sketch itself ? Otherwise, it sounds pretty much the same as Pad (including optional taper, for instance).

mno13, 2024-05-26:
Yes, i would want to edit hole parameters without editing sketch. Hole could also have tapered end (from drill) and have counter-bore or countersunk.  On editing sketch - hole tool needing circles, not points for positioning holes is not so good solution, as it uses only circle centres (which are itself points), so there is useless geometry that "may" be left unconstrained (wrong practice) or given diameter not corresponding to hole diameter (again, wrong practice).

Tool for boss feature is in every mid-range cad package, solid edge, solidworks, inventor etc,

CATIA have ability to create "positive hole" and use it as tool in boolean op.