FreeCAD/FreeCAD #14283: BIM: Error when entering measurements to walls before creating them

gianlucafiore, 2024-05-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
[image]
The steps to follow are: 
- Click the Wall tool to start drawing one from scratch. 
- Enter wall height and width measurements.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37464 (Git)
Build type: Release
Branch: main
Hash: db48a78d5a545bb21a5c80a9fa4b4c983fbbe951
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-26:
Can confirm.
@yorikvanhavre FYI

yorikvanhavre, 2024-05-27:
Confirmed too. Preparing a PR

PR #14315 (2024-05-28): BIM: Fixed wall paramerers
Fixes #14283
commit f07857e8 (2024-05-28): BIM: Fixed wall paramerers - fixes #14283 (#14315)
closed 2024-05-28 by Roy-043