FreeCAD/FreeCAD #14287: FreeCAD's interface freezes on linux wayland session

oficinerobotica, 2024-05-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While using the weekly builds downloaded from here https://github.com/FreeCAD/FreeCAD-Bundle/releases
or Ondsel downloaded from here https://github.com/Ondsel-Development/FreeCAD/releases/

Applying any kind of theme will freeze the interface or, for a better explanation will slow it down to the point of having up to 60 seconds of delay between a click on a interface item and a response from the app. I'm using appimage on kde neon 6.0 and a message that appears if I start in a terminal is this: 

`(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""`

in my case, on a wayland session, using appimage, with clean config folders freecad is usable as long as I don't make the initial theme selection in the start page. I have to ignore it. As soon as I chose a theme in the start page the interface becomes unresponsive but doesn't crash. It takes like 60 seconds between a click and a response from the interface. The only way to revert to a usable state is to delete the config folders. 

This behavior dissapears if I switch to a xserver session 

Forum post [https://forum.freecad.org/viewtopic.php?t=87325](url)

### Full version info

```shell
OS: KDE neon 6.0 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37083 (Git) AppImage
Build type: Release
Branch: main
Hash: 502f78e111c761cb92c51222605b26b8701fcbca
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Installed mods: 
  * fasteners 0.5.17
  * Behave-Dark-Colors 0.1.1
  * Curves 0.6.24
  * Assembly3 0.12.2
  * CurvedShapes 1.0.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-05-26:
Have you tried to install `qtwayland5` as proposed here?[^1]

Is there a _Qt_ software that behaves in the same way?

[^1]: https://stackoverflow.com/questions/69994530/qt-qpa-plugin-could-not-find-the-qt-platform-plugin-wayland

wwmayer, 2024-05-26:
According to `https://github.com/qutebrowser/qutebrowser/issues/5491` the issue might be caused by `QT_WAYLAND_FORCE_DPI=physical`. Is this is environment variable set on your system and if yes does unsetting it solve the issue?

oficinerobotica, 2024-05-26:
After reading the post above I have controlled and I have both `qtwaylan5` and `qt6-wayland` to the latest releases on my system. No other qt app is giving me problems so far. In the post above obs studio is mentioned, I have it installed and it works correctly. What can I verify to help trace down the problem?

maxwxyz, 2024-05-26:
In https://github.com/FreeCAD/FreeCAD/issues/6177 someone mentioned that the Appimage wouldn't support wayland.

Could this be related to https://github.com/FreeCAD/FreeCAD/issues/13893 or https://github.com/FreeCAD/FreeCAD/issues/12622 ?

oficinerobotica, 2024-05-26:
@wwmayer sorry, I somehow missed your reply .... how can I verify that on my system?

wwmayer, 2024-05-26:
Open a bash and enter: `echo $QT_WAYLAND_FORCE_DPI`.

If it's set then you can unset it with `export $QT_WAYLAND_FORCE_DPI=` for this running instance of the bash. From within there you can start FreeCAD and check if it solves the issue or not.

oficinerobotica, 2024-05-26:
`flo@flo-cyborg15a13ve:~$ echo $QT_WAYLAND_FORCE_DPI

flo@flo-cyborg15a13ve:~$ export $QT_WAYLAND_FORCE_DPI=
bash: export: `=': not a valid identifier
flo@flo-cyborg15a13ve:~$ 
`

Unfortunately neither of the commands give a result on my system.

wwmayer, 2024-05-26:
> echo $QT_WAYLAND_FORCE_DPI

It's not set.

> export $QT_WAYLAND_FORCE_DPI=

Oops, may bad. This should be `export QT_WAYLAND_FORCE_DPI=` but as it's not set this doesn't change anything.

obelisk79, 2024-05-26:
On an Arch based system I just had an update from Qt 6.7.0 > 6.7.1 and KDE/Plasma to version 6.0.5-1 and the freezes have resolved on my system. I've been struggling to deal with freezes as reported above for a few weeks.

kinghat, 2024-05-26:
im still having the issue on 6.0.5 unfortunately. i clear the .local/.cache/.config dirs before each run as well:

> OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.37464 (Git) AppImage
> Build type: Release
> Branch: main
> Hash: db48a78d5a545bb21a5c80a9fa4b4c983fbbe951
> Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
> Locale: English/United States (en_US)

> Operating System: Fedora Linux 40
> KDE Plasma Version: 6.0.5
> KDE Frameworks Version: 6.2.0
> Qt Version: 6.7.0
> Kernel Version: 6.8.10-300.fc40.x86_64 (64-bit)
> Graphics Platform: Wayland
> Processors: 4 × Intel® Core™ i7-5600U CPU @ 2.60GHz
> Memory: 11.6 GiB of RAM
> Graphics Processor: Mesa Intel® HD Graphics 5500

obelisk79, 2024-05-26:
@kinghat Look for a qt upgrade from 6.7.0 to 6.7.1.

oficinerobotica, 2024-05-28:
For some reason, on my system, after the latest system update all the freezing has disappeared. And that is a good news :)
I don't really know what changed tough.My system is still on qt 6.7.0
Should this bug report be closed?

krushia, 2024-05-29:
The themes are really slow to load in general. Launching FreeCAD will take up to 20 seconds longer with a theme selected vs. just using Classic. It also "freezes" for about 10 seconds when changing stylesheets in preferences.

I assume this bug is about a much more horrible condition?

obelisk79, 2024-05-29:
@krushia I believe this is related to whatever was causing the bug in this report. Recent Qt/KDE updates have resolved this for the most part so depending on whatever linux distribution you use you'll need to wait for the updated libraries most likely.

oficinerobotica, 2024-06-01:
In a previous comment I said that the problem is gone but unfortunately that is not the case. This is what I see when opening FC. Still waiting for the qt 6.7.1 on my system to see if it changes something

[image]

kinghat, 2024-06-20:
ive said this before but the fedora plasma 6.1 update ive not had the freezing issues on freecad and ondsel:

> Operating System: Fedora Linux 40
> KDE Plasma Version: 6.1.0
> KDE Frameworks Version: 6.3.0
> Qt Version: 6.7.1
> Kernel Version: 6.9.4-200.fc40.x86_64 (64-bit)
> Graphics Platform: Wayland
> Processors: 4 × Intel® Core™ i7-5600U CPU @ 2.60GHz
> Memory: 11.6 GiB of RAM
> Graphics Processor: Mesa Intel® HD Graphics 5500

maxwxyz, 2024-07-15:
We'll close this in favor of https://github.com/FreeCAD/FreeCAD/issues/15278 and https://github.com/FreeCAD/FreeCAD/issues/15132 as this issue seems to be a combination of both. If fixes of these issues do not resolve the issue, please let us know.

closed 2024-07-15 by maxwxyz

JannikWilhelm, 2024-07-18:
I am having the same issue on Fedora KDE. When Switching over to gnome, these issues disappear.

nuggiz, 2024-09-18:
Plasma session on my system had set 
QT_WAYLAND_RECONNECT=1
QT_AUTO_SCREEN_SCALE_FACTOR=0
unsetting them seems to fix it the freezes.
OS: Arch Linux x86_64
DE: Plasma 6.1.5 Wayland

evranch, 2024-09-21:
Just switched my Manjaro system to Wayland for better support of VRR and this bug reared its head. Completely unusable under Wayland.

All packages up to date and was working great under X.
Unsetting those environment variables has no effect.

luzpaz, 2024-09-22:
@evranch can you please add your full [About](https://wiki.freecad.org/About) info?

evranch, 2024-09-24:
I'm running Ondsel. Had to launch it under X instead of Wayland to get to the About screen.

Oof, just realized that there was a bugfix released, 2024.2.2 that I didn't see in AUR because Ondsel decided to hide it behind a mandatory login. Problem solved, though I'm not impressed by the change in distribution method. Sorry for the stale report.

OS: Manjaro Linux (KDE/plasma)
Word size of Ondsel: 64-bit
Version: 2024.2.1.37226 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 2024.2.1)
Hash: 1a1bd7290494aef925a026c87b808ac58d53c3b0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Ondsel-Lens 2024.1.22.01

jakemagee, 2024-11-13:
Seeing the same issue here with 1.0.0 RC2 and weekly build 39176.  Issue is resolved when switching to X11 session from Wayland.  Interestingly, it seems a "refresh" / "unfreeze" can be somewhat temporarily forced by moving the mouse back and forth to other monitors / windows.

OS: Ubuntu 24.10 (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.1.0dev.39176 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 45d3864942c6928cec7dc18f95b1f222485756e5
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default