FreeCAD/FreeCAD #14298: CAM: preference card needs revision [AppImage]

kaktusus, 2024-05-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

if you use a language other than English in the GUI, the layout of the text is incorrect. The text overlaps the boxes.

https://forum.freecad.org/viewtopic.php?style=4&t=87357

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37464 (Git) AppImage
Build type: Release
Branch: main
Hash: db48a78d5a545bb21a5c80a9fa4b4c983fbbe951
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Installed mods: 
...
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-27:
There seems to be a lot of white space to the right, is this only for the CAM settings?

kaktusus, 2024-05-27:
Yes, and only this one preference card.
I have not noticed such an anomaly in the other workbenches.

maxwxyz, 2024-05-27:
@sliptonic FYI

xtemp09, 2024-05-28:
I cannot reproduce.

<details><summary>Polska</summary>

[image]

[image]

</details>

<details><summary>Portuguesa</summary>

[image]

</details>

<details><summary>Ubuntu 22.04.3 LTS (KDE/plasma)</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37464 (Git)
Build type: Debug
Branch: main
Hash: db48a78d5a545bb21a5c80a9fa4b4c983fbbe951
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Polish/Poland (pl_PL)
```

</details>

@kaktusus, do stable and oldstable versions of _FreeCAD_ show the same behavior?

kaktusus, 2024-05-28:
Yes this "design" I have had for a long time.
I do not use the stable version.

kaktusus, 2024-05-28:
problem solved:
https://github.com/FreeCAD/FreeCAD/pull/14343

kaktusus, 2024-05-28:
xtemp09 
thank you for your efforts.

closed 2024-05-28 by kaktusus