FreeCAD/FreeCAD #14307: Undefined symbol/Checking for Connection to Github

iplayfast, 2024-05-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

02:03:14  During initialization the error "/home/chris/.local/lib/python3.10/site-packages/PySide2/QtUiTools.abi3.so: undefined symbol: _ZdlPvm, version Qt_5" occurred in /home/chris/.local/share/FreeCAD/Mod/btl/./InitGui.py

I'm also finding that when I click on the add on manager, it says that it is "Checking for Connection to githu" and just sits there. I don't know if it is related or not.

### Full version info

```shell
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37472 (Git)
Build type: Unknown
Branch: main
Hash: d4f780782dae8d849caf30ae65885e32ff3fa635
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/Canada (en_CA)
Installed mods: 
  * freecad.gears 1.0.0
  * 3D_Printing_Tools
  * btl 0.9.9
  * CycloidGearBox
  * FeedsAndSpeeds 0.5.0
  * Ondsel-Lens 2024.1.22.01
  * ExplodedAssembly
  * Nodes 0.1.36
  * 3DfindIT 1.2.0
  * A2plus 0.4.63
  * MeshRemodel 1.9.22
  * lattice2 1.0.0
  * fasteners 0.5.10
  * Assembly3 0.12.2
  * Assembly4 0.50.7
  * POV-Ray-Rendering
  * PieMenu 1.2.7
  * CurvedShapes 1.0.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-27:
Seems like an issue in the blt addon. Can you remove this folder under Mod/ and try again?

iplayfast, 2024-05-27:
That got rid of the 02:03:14 During initialization the error "/home/chris/.local/lib/python3.10/site-packages/PySide2/QtUiTools.abi3.so: undefined symbol: _ZdlPvm, version Qt_5" occurred in /home/chris/.local/share/FreeCAD/Mod/btl/./InitGui.py problem.

Still have the  "Checking for Connection to github"  dialog just sitting there.

maxwxyz, 2024-05-27:
@chennes FYI, it seems to get no connection when launching the addon manager

macdroid53, 2024-06-07:
Not sure if this is the same issue, but this showed up yesterday and today on Manjaro. Gets to this point and hangs indefinitely.
[image]

Jookia, 2024-07-01:
Is this fixed in dev?

chennes, 2024-07-01:
Yes (at least, it should be!).

maxwxyz, 2024-07-01:
So this issue can be closed?

iplayfast, 2024-07-12:
As of July 12, 2024 it's not fixed.

chennes, 2024-07-12:
Then we need to start to gather more information (the connection check failing is a symptom of many different unrelated problems). Can you please turn logging on and paste in the output?

iplayfast, 2024-07-12:
I've been doing git bisect but have not found it. 
How do you turn on logging?

iplayfast, 2024-07-12:
[log.txt](https://github.com/user-attachments/files/16199755/log.txt)

luzpaz, 2024-07-18:
Is this related to #15059

chennes, 2024-07-19:
> Is this related to #15059 ?

No -- that was a Qt6 issue.

chennes, 2024-07-19:
> [log.txt](https://github.com/user-attachments/files/16199755/log.txt)

Well, that's a dead end, there's nothing at all in there. Can you try, from the FreeCAD Python Console:
```
import NetworkManager
NetworkManager.InitializeNetworkManager()
NetworkManager.AM_NETWORK_MANAGER.blocking_get("https://api.github.com/zen")
```

iplayfast, 2024-07-20:
From the menu
```
Python 3.10.12 (main, Mar 22 2024, 16:50:05) [GCC 11.4.0] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> # Gui.runCommand('Start_Start',0)
>>> # Gui.runCommand('Std_Workbench',21)
>>> # Gui.runCommand('Std_Workbench',21)
>>> # Gui.runCommand('Std_ViewStatusBar',1) < clicked on addon manager here
>>> # Gui.runCommand('Std_AddonMgr',0)  < stalls here
```

Trying your commands
```
import NetworkManager
NetworkManager.InitializeNetworkManager()
NetworkManager.AM_NETWORK_MANAGER.blocking_get("https://api.github.com/zen") < stalls here
```
If I push enter again it says 
```
Previous command still running!
```
ctrl-c gives 
```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/chris/github/FreeCAD/build/Mod/AddonManager/NetworkManager.py", line 396, in blocking_get
    QtCore.QCoreApplication.processEvents()
KeyboardInterrupt
```

iplayfast, 2024-08-02:
ping!

chennes, 2024-08-02:
Based on your console results, it looks like a Qt bug -- where are you getting the executable?

iplayfast, 2024-08-05:
I'm compiling them from github. running Linux Mint
```
chris@FORGE:~/github/FreeCAD/build/bin$ ldd FreeCAD
	linux-vdso.so.1 (0x00007fff7adf7000)
	libFreeCADGui.so => /home/chris/github/FreeCAD/build/lib/libFreeCADGui.so (0x00007ba9eee00000)
	libFreeCADApp.so => /home/chris/github/FreeCAD/build/lib/libFreeCADApp.so (0x00007ba9eda00000)
	libFreeCADBase.so => /home/chris/github/FreeCAD/build/lib/libFreeCADBase.so (0x00007ba9ed400000)
	libpython3.10.so.1.0 => /lib/x86_64-linux-gnu/libpython3.10.so.1.0 (0x00007ba9ece00000)
	libfmt.so.8 => /lib/x86_64-linux-gnu/libfmt.so.8 (0x00007ba9ed9df000)
	libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007ba9ec600000)
	libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007ba9ebe00000)
	libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007ba9eb800000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007ba9eb400000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007ba9ed9bf000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ba9eb000000)
	libCoin.so.80c => /lib/x86_64-linux-gnu/libCoin.so.80c (0x00007ba9ea600000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007ba9ed938000)
	libQt5PrintSupport.so.5 => /lib/x86_64-linux-gnu/libQt5PrintSupport.so.5 (0x00007ba9ecd88000)
	libQt5Svg.so.5 => /lib/x86_64-linux-gnu/libQt5Svg.so.5 (0x00007ba9ecd2b000)
	libQt5Network.so.5 => /lib/x86_64-linux-gnu/libQt5Network.so.5 (0x00007ba9eb64f000)
	libspnav.so.0 => /lib/libspnav.so.0 (0x00007ba9ea200000)
	libpyside2.cpython-310-x86_64-linux-gnu.so.5.15 => /lib/x86_64-linux-gnu/libpyside2.cpython-310-x86_64-linux-gnu.so.5.15 (0x00007ba9eccfa000)
	libboost_filesystem.so.1.74.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.74.0 (0x00007ba9ed3e0000)
	libboost_regex.so.1.74.0 => /lib/x86_64-linux-gnu/libboost_regex.so.1.74.0 (0x00007ba9ec50d000)
	libxerces-c-3.2.so => /lib/x86_64-linux-gnu/libxerces-c-3.2.so (0x00007ba9e9e00000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007ba9eb319000)
	libQt5Xml.so.5 => /lib/x86_64-linux-gnu/libQt5Xml.so.5 (0x00007ba9ebdba000)
	libshiboken2.cpython-310-x86_64-linux-gnu.so.5.15 => /lib/x86_64-linux-gnu/libshiboken2.cpython-310-x86_64-linux-gnu.so.5.15 (0x00007ba9ebd6e000)
	/lib64/ld-linux-x86-64.so.2 (0x00007ba9f1b1e000)
	libboost_program_options.so.1.74.0 => /lib/x86_64-linux-gnu/libboost_program_options.so.1.74.0 (0x00007ba9eb2d4000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007ba9eccde000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007ba9eb2a3000)
	libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007ba9eb268000)
	libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007ba9eaf31000)
	libmd4c.so.0 => /lib/x86_64-linux-gnu/libmd4c.so.0 (0x00007ba9eedac000)
	libdouble-conversion.so.3 => /lib/x86_64-linux-gnu/libdouble-conversion.so.3 (0x00007ba9eccc9000)
	libicui18n.so.70 => /lib/x86_64-linux-gnu/libicui18n.so.70 (0x00007ba9e9a00000)
	libicuuc.so.70 => /lib/x86_64-linux-gnu/libicuuc.so.70 (0x00007ba9ea405000)
	libpcre2-16.so.0 => /lib/x86_64-linux-gnu/libpcre2-16.so.0 (0x00007ba9eaea7000)
	libzstd.so.1 => /lib/x86_64-linux-gnu/libzstd.so.1 (0x00007ba9e9d31000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007ba9e98c6000)
	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007ba9eb234000)
	libOpenGL.so.0 => /lib/x86_64-linux-gnu/libOpenGL.so.0 (0x00007ba9ec4e1000)
	libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007ba9e980e000)
	libgssapi_krb5.so.2 => /lib/x86_64-linux-gnu/libgssapi_krb5.so.2 (0x00007ba9eae53000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007ba9e96ce000)
	libQt5Qml.so.5 => /lib/x86_64-linux-gnu/libQt5Qml.so.5 (0x00007ba9e9200000)
	libcurl-gnutls.so.4 => /lib/x86_64-linux-gnu/libcurl-gnutls.so.4 (0x00007ba9ea15e000)
	libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007ba9e9138000)
	libgraphite2.so.3 => /lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007ba9e96a7000)
	libicudata.so.70 => /lib/x86_64-linux-gnu/libicudata.so.70 (0x00007ba9e7400000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007ba9e90c2000)
	libkrb5.so.3 => /lib/x86_64-linux-gnu/libkrb5.so.3 (0x00007ba9e7335000)
	libk5crypto.so.3 => /lib/x86_64-linux-gnu/libk5crypto.so.3 (0x00007ba9e9093000)
	libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x00007ba9f1aa8000)
	libkrb5support.so.0 => /lib/x86_64-linux-gnu/libkrb5support.so.0 (0x00007ba9ebd60000)
	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007ba9e9069000)
	libnghttp2.so.14 => /lib/x86_64-linux-gnu/libnghttp2.so.14 (0x00007ba9e903f000)
	libidn2.so.0 => /lib/x86_64-linux-gnu/libidn2.so.0 (0x00007ba9eb62e000)
	librtmp.so.1 => /lib/x86_64-linux-gnu/librtmp.so.1 (0x00007ba9e9020000)
	libssh.so.4 => /lib/x86_64-linux-gnu/libssh.so.4 (0x00007ba9e72c8000)
	libpsl.so.5 => /lib/x86_64-linux-gnu/libpsl.so.5 (0x00007ba9e9693000)
	libnettle.so.8 => /lib/x86_64-linux-gnu/libnettle.so.8 (0x00007ba9e7282000)
	libgnutls.so.30 => /lib/x86_64-linux-gnu/libgnutls.so.30 (0x00007ba9e7097000)
	libldap-2.5.so.0 => /lib/x86_64-linux-gnu/libldap-2.5.so.0 (0x00007ba9e7038000)
	liblber-2.5.so.0 => /lib/x86_64-linux-gnu/liblber-2.5.so.0 (0x00007ba9e7027000)
	libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x00007ba9e7019000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007ba9ed931000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007ba9e7005000)
[... cut ...]
	libp11-kit.so.0 => /lib/x86_64-linux-gnu/libp11-kit.so.0 (0x00007ba9e6c56000)
	libtasn1.so.6 => /lib/x86_64-linux-gnu/libtasn1.so.6 (0x00007ba9e67e8000)
	libsasl2.so.2 => /lib/x86_64-linux-gnu/libsasl2.so.2 (0x00007ba9e67cd000)
	libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x00007ba9e67aa000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007ba9e6792000)
	libffi.so.8 => /lib/x86_64-linux-gnu/libffi.so.8 (0x00007ba9eae46000)
	libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x00007ba9e6c49000)
```

iplayfast, 2024-08-06:
Interesting. I just downloaded the freecad appimage https://github.com/FreeCAD/FreeCAD/releases/download/0.21.2/FreeCAD-0.21.2-Linux-x86_64.AppImage and it has the same issue. This one doesn't use local libraries at all so I think we can rule out qt error, although if it works for others, I can't image what it could be.

chennes, 2024-08-07:
Well, it's presumably some odd interaction between Qt and your particular networking stack. The fact that Ondsel works is very interesting, what are the various Qt versions involved with Ondsel, the 0.21.2 appimage, and your local compile?

iplayfast, 2024-08-07:
Well as they are app images (ie statically linked) I have no idea. 
Ondsel is an app image and works, Freecad app image doesn't. My compile doesn't. All I can show you is my compile stuff (above). 
If it's working for you, what does yours say when you do an ldd of FreeCAD ?

ClayShoaf, 2024-08-08:
> Interesting. I just downloaded the freecad appimage https://github.com/FreeCAD/FreeCAD/releases/download/0.21.2/FreeCAD-0.21.2-Linux-x86_64.AppImage and it has the same issue. This one doesn't use [...]

The AppImage is not working for me either. 
Python console output:

```
Network Manager Error: __reply_finished not called by a Qt signal
```

Tried with a native install via apt and the output has another reference to Qt:

```
17:08:42  Addon Manager Warning: Could not import QtWebEngineWidgets -- README data will display as text-only
17:08:45  The `.FCStd#` backup format is deprecated as of v0.21 and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
17:09:00  Network Manager Error: __reply_finished not called by a Qt signal
```

Ubuntu 22.04

iplayfast, 2024-08-11:
I never thought to try installing with apt. (so old!) but I tried it and it worked! So here are the libraries linked in with it.
```
ldd /usr/bin/freecad
	linux-vdso.so.1 (0x00007ffd5b5cf000)
	libFreeCADGui.so => /usr/lib/freecad-python3/lib/libFreeCADGui.so (0x0000787b52200000)
	libFreeCADApp.so => /usr/lib/freecad-python3/lib/libFreeCADApp.so (0x0000787b51e00000)
	libFreeCADBase.so => /usr/lib/freecad-python3/lib/libFreeCADBase.so (0x0000787b5315f000)
	libpython3.10.so.1.0 => /lib/x86_64-linux-gnu/libpython3.10.so.1.0 (0x0000787b51600000)
	libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x0000787b50e00000)
	libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x0000787b50600000)
	libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x0000787b50000000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x0000787b4fc00000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x0000787b5313d000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x0000787b4f800000)
	libCoin.so.80c => /lib/x86_64-linux-gnu/libCoin.so.80c (0x0000787b4ee00000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x0000787b51d37000)
	libQt5PrintSupport.so.5 => /lib/x86_64-linux-gnu/libQt5PrintSupport.so.5 (0x0000787b51cbf000)
	libQt5Svg.so.5 => /lib/x86_64-linux-gnu/libQt5Svg.so.5 (0x0000787b51c62000)
	libQt5Network.so.5 => /lib/x86_64-linux-gnu/libQt5Network.so.5 (0x0000787b4fe4f000)
	libspnav.so.0 => /lib/libspnav.so.0 (0x0000787b4ea00000)
	libpyside2.cpython-310-x86_64-linux-gnu.so.5.15 => /lib/x86_64-linux-gnu/libpyside2.cpython-310-x86_64-linux-gnu.so.5.15 (0x0000787b51c31000)
	libboost_regex.so.1.74.0 => /lib/x86_64-linux-gnu/libboost_regex.so.1.74.0 (0x0000787b5150d000)
	libxerces-c-3.2.so => /lib/x86_64-linux-gnu/libxerces-c-3.2.so (0x0000787b4e600000)
	libzipios.so.0 => /lib/x86_64-linux-gnu/libzipios.so.0 (0x0000787b51c00000)
	libQt5Xml.so.5 => /lib/x86_64-linux-gnu/libQt5Xml.so.5 (0x0000787b514c7000)
	libshiboken2.cpython-310-x86_64-linux-gnu.so.5.15 => /lib/x86_64-linux-gnu/libshiboken2.cpython-310-x86_64-linux-gnu.so.5.15 (0x0000787b50db4000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x0000787b4fb19000)
	/lib64/ld-linux-x86-64.so.2 (0x0000787b532ef000)
	libboost_program_options.so.1.74.0 => /lib/x86_64-linux-gnu/libboost_program_options.so.1.74.0 (0x0000787b50d6f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x0000787b51be4000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x0000787b50d3e000)
	libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x0000787b50d03000)
	libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x0000787b4fa4a000)
	libmd4c.so.0 => /lib/x86_64-linux-gnu/libmd4c.so.0 (0x0000787b50cf1000)
	libdouble-conversion.so.3 => /lib/x86_64-linux-gnu/libdouble-conversion.so.3 (0x0000787b505eb000)
	libicui18n.so.70 => /lib/x86_64-linux-gnu/libicui18n.so.70 (0x0000787b4e200000)
	libicuuc.so.70 => /lib/x86_64-linux-gnu/libicuuc.so.70 (0x0000787b4ec05000)
	libpcre2-16.so.0 => /lib/x86_64-linux-gnu/libpcre2-16.so.0 (0x0000787b50561000)
	libzstd.so.1 => /lib/x86_64-linux-gnu/libzstd.so.1 (0x0000787b4f731000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x0000787b4e0c6000)
	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x0000787b4f6fd000)
	libOpenGL.so.0 => /lib/x86_64-linux-gnu/libOpenGL.so.0 (0x0000787b4f6d1000)
	libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x0000787b4e548000)
	libgssapi_krb5.so.2 => /lib/x86_64-linux-gnu/libgssapi_krb5.so.2 (0x0000787b4f67d000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x0000787b4df86000)
	libQt5Qml.so.5 => /lib/x86_64-linux-gnu/libQt5Qml.so.5 (0x0000787b4da00000)
	libcurl-gnutls.so.4 => /lib/x86_64-linux-gnu/libcurl-gnutls.so.4 (0x0000787b4e95e000)
	libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x0000787b4debe000)
	libgraphite2.so.3 => /lib/x86_64-linux-gnu/libgraphite2.so.3 (0x0000787b4f656000)
	libicudata.so.70 => /lib/x86_64-linux-gnu/libicudata.so.70 (0x0000787b4bc00000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x0000787b4d98a000)
	libkrb5.so.3 => /lib/x86_64-linux-gnu/libkrb5.so.3 (0x0000787b4d8bf000)
	libk5crypto.so.3 => /lib/x86_64-linux-gnu/libk5crypto.so.3 (0x0000787b4de8f000)
	libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x0000787b521f4000)
	libkrb5support.so.0 => /lib/x86_64-linux-gnu/libkrb5support.so.0 (0x0000787b50ce3000)
	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x0000787b4d895000)
	libnghttp2.so.14 => /lib/x86_64-linux-gnu/libnghttp2.so.14 (0x0000787b4d86b000)
	libidn2.so.0 => /lib/x86_64-linux-gnu/libidn2.so.0 (0x0000787b4fe2e000)
	librtmp.so.1 => /lib/x86_64-linux-gnu/librtmp.so.1 (0x0000787b4fa2b000)
	libssh.so.4 => /lib/x86_64-linux-gnu/libssh.so.4 (0x0000787b4bb93000)
	libpsl.so.5 => /lib/x86_64-linux-gnu/libpsl.so.5 (0x0000787b4e534000)
	libnettle.so.8 => /lib/x86_64-linux-gnu/libnettle.so.8 (0x0000787b4d825000)
	libgnutls.so.30 => /lib/x86_64-linux-gnu/libgnutls.so.30 (0x0000787b4b9a8000)
	libldap-2.5.so.0 => /lib/x86_64-linux-gnu/libldap-2.5.so.0 (0x0000787b4b949000)
	liblber-2.5.so.0 => /lib/x86_64-linux-gnu/liblber-2.5.so.0 (0x0000787b4b938000)
	libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x0000787b4f648000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x0000787b51bd7000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x0000787b4b924000)
	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x0000787b514c1000)
[... cut ...]
	libp11-kit.so.0 => /lib/x86_64-linux-gnu/libp11-kit.so.0 (0x0000787b4b0c5000)
	libtasn1.so.6 => /lib/x86_64-linux-gnu/libtasn1.so.6 (0x0000787b4b698000)
	libsasl2.so.2 => /lib/x86_64-linux-gnu/libsasl2.so.2 (0x0000787b4b67d000)
	libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x0000787b4b65a000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x0000787b4b0ad000)
	libffi.so.8 => /lib/x86_64-linux-gnu/libffi.so.8 (0x0000787b4b64d000)
	libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x0000787b4b0a0000)
```

iplayfast, 2024-08-11:
Comparing libraries linked by both, they are the same except that the compiled version includes libboost filesystem.so.1.74.0 and libfmt.so.8
The apt install version doesn't have those, but  includes libzipios.so.0
Hopefully a clue.

chennes, 2024-08-11:
The version information you get from ldd isn't enough, I'd like to know the precise versions of Qt that are working versus not-working. Can you post the "About FreeCAD" info from these?

iplayfast, 2024-08-11:
Working on Bisecting git, 
Earlier versions fail, but have this message
Network Manager Error: __reply_finished not called by a Qt signal
Even earlier versions fail, but have this additional message
ddon Manager Warning: Could not import QtWebEngineWidgets, it seems to be missing from your system. Please use your system's package manager to install the python3-pyside2.qtwebengine* and python3-pyside2.qtwebchannel packages, and if possible alert your package creator to the missing dependency. Display of package README will be limited until this dependency is resolved.
0.89916 <asm3.main> init_gui.py(14): no solver backend found
(QtWebEngineWidgets is installed on my system)
==============
The info you asked for.
I did a totally fresh rebuild (git clone --recurse-submodules https://github.com/FreeCAD/FreeCAD.git)

OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38432 (Git)
Build type: Unknown
Branch: main
Hash: b5281158f30732987bfa15a03b66928be7b41529
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/Canada (en_CA)
Installed mods: 
  * freecad.gears 1.0.0
  * 3D_Printing_Tools
  * CycloidGearBox
  * FeedsAndSpeeds 0.5.0
  * Ondsel-Lens 2024.1.22.01
  * ExplodedAssembly
  * Nodes 0.1.36
  * 3DfindIT 1.2.0
  * A2plus 0.4.63
  * MeshRemodel 1.9.22
  * lattice2 1.0.0
  * fasteners 0.5.10
  * Assembly3 0.12.2
  * Assembly4 0.50.7
  * POV-Ray-Rendering
  * PieMenu 1.2.7
  * CurvedShapes 1.0.5
From the about/libraries tab:
This software uses open source components whose copyright and other proprietary rights belong to their respective owners:

Boost 1_74
[https://www.boost.org](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocBoost)

Coin3D 4.0.0
[https://coin3d.github.io](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocCoin3D)

Eigen 3.4.0
[https://eigen.tuxfamily.org](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocEigen)

FreeType 2.11.1
[https://freetype.org](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocFreeType)

KDL 
[https://www.orocos.org/kdl](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocKDL)

libarea 
[https://github.com/danielfalck/libarea](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocLibArea)

Open CASCADE Technology 7.5.1
[https://www.opencascade.com/open-cascade-technology/](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocOCCT)

Point Cloud Library 
[https://www.pointclouds.org](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocPcl)

PyCXX 7.1.9
[http://cxx.sourceforge.net](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocPyCXX)

Python 3.10.12
[https://www.python.org](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocPython)

Qt for Python (PySide) 5.15.2
[https://wiki.qt.io/Qt_for_Python](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocPySide)

Qt 5.15.3
[https://www.qt.io](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocQt)

Salome SMESH 7.7.1.0
[https://salome-platform.org](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocSalomeSMESH)

Qt for Python (Shiboken) 5.15.2
[https://wiki.qt.io/Qt_for_Python](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocPySide)

vtk 9.1.0
[https://www.vtk.org](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocVtk)

Xerces-C 3.2.3
[https://xerces.apache.org/xerces-c](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocXercesC)

Zipios++ 
[http://zipios.sourceforge.net](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocZipios)

zlib 1.2.11
[https://zlib.net](file:///home/chris/github/FreeCAD/build/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocZlib)

chennes, 2024-08-12:
Is 5.15.3 the latest version of Qt available on Mint 21.3?

iplayfast, 2024-08-12:
Yes, Or I could install  6.2.4, but I think the 5 is the standard one.and I don't want to break packages.
It's strange that it used to work, but now doesn't.

iplayfast, 2024-08-13:
I tried editing the file, but could not figure out where to put it.

iplayfast, 2024-09-01:
Should this issue be considered a blocker, since anyone with an up to date ubuntu or linux mint will have this problem. It appears to be a qt 3.15.3 vs qt 3.15.2 problem but it is a major problem for getting updates.

chennes, 2024-09-01:
No one else has reported the issue -- I have tested on Ubuntu 20.04, 22.04, and 24.04. It appears to be a combination of a particular version of Qt and some particular hardware configuration. Considering that there is always a workaround (addons can be installed by downloading the Zip, or by using git clone on the command line), this does not block the release of 1.0.

iplayfast, 2024-09-01:
@ClayShoaf also said that he was having the same problem earlier in the thread. I know I can work around it, I just would hate for the release to have this problem with many unsuspecting new users. The versions that you tested on, were any of them qt.3.15.3?

chennes, 2024-09-01:
@ClayShoaf was testing 0.21.2, which has a number of known bugs in this part of the networking code. 

I'm open to suggestions for how to debug or work around the problem: the core of the Addon Manager networking stack is simply stalling out on your machine, not only failing to return the expected data, but both failing to report a connection failure, and failing to timeout. We're getting exactly zero diagnostic output. There's not much to go on here. You could try adding some `print` statements in NetworkManager.py to try to isolate the actual Qt call that is failing. There's a chance that it's something that could be coded around.

iplayfast, 2024-09-02:
I added tracking to Networkmanager.py and addonmanager.py
which should track any blocking getmessage. 
Now when I start freecad and try to updateaddons, I get the following
```
reecad
FreeCAD 0.22.0, Libs: 0.22.0devR38622 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

connect failed: No such file or directory
AM_NETWORK_MANAGER is None
hello
No blocking requests
connected should be True
No blocking requests
```
AM_NETOWRK_MANAGER is None on initiailzation (which is expected)
Hello is the start of the getting addons
```
  def start(self):
        """Start the connection check"""
        print("hello");
        status = NetworkManager.AM_NETWORK_MANAGER.get_status()
        print(status)  # Outputs "Blocked by request to [URL]" or "No blocking requests"
        self.connection_checker.start()
        self.connection_checker.success.connect(self._check_succeeded)
        self.connection_checker.failure.connect(self._network_connection_failed)
        self.signals_connected = True
        print("connected should be True")
        status = NetworkManager.AM_NETWORK_MANAGER.get_status()
        print(status)  # Outputs "Blocked by request to [URL]" or "No blocking requests"
        # If it takes longer than a half second to check the connection, show a message:
        self.connection_message_timer = QtCore.QTimer.singleShot(
            1500, self._show_connection_check_message
        )
```
So it looks like there are no blocking requests but the connection is not working, and does not timeout after 5 minutes. 

running again says
```
AM_NETWORK_MANAGER already initialized
hello
No blocking requests
connected should be True
No blocking requests
```
So the first request did not block eithe, or was canceled when I cancelled the dialog.

If you can think of anything else to try, I'm game.

Code I modified is below, 

```
diff --git a/src/Mod/AddonManager/NetworkManager.py b/src/Mod/AddonManager/NetworkManager.py
index 3d391e670b..10fa9fe82e 100644
--- a/src/Mod/AddonManager/NetworkManager.py
+++ b/src/Mod/AddonManager/NetworkManager.py
@@ -163,6 +163,9 @@ if HAVE_QTNETWORK:
             self.synchronous_complete: Dict[int, bool] = {}
             self.synchronous_result_data: Dict[int, QtCore.QByteArray] = {}
 
+            # Added to assist in debugging
[... 57 lines cut ...]
--- a/src/Mod/AddonManager/addonmanager_connection_checker.py
+++ b/src/Mod/AddonManager/addonmanager_connection_checker.py
@@ -52,14 +52,19 @@ class ConnectionCheckerGUI(QtCore.QObject):
 
     def start(self):
         """Start the connection check"""
+        print("hello");
+        status = NetworkManager.AM_NETWORK_MANAGER.get_status()
+        print(status)  # Outputs "Blocked by request to [URL]" or "No blocking requests"
         self.connection_checker.start()
         self.connection_checker.success.connect(self._check_succeeded)
         self.connection_checker.failure.connect(self._network_connection_failed)
         self.signals_connected = True
-
+        print("connected should be True")
+        status = NetworkManager.AM_NETWORK_MANAGER.get_status()
+        print(status)  # Outputs "Blocked by request to [URL]" or "No blocking requests"
         # If it takes longer than a half second to check the connection, show a message:
         self.connection_message_timer = QtCore.QTimer.singleShot(
-            500, self._show_connection_check_message
+            1500, self._show_connection_check_message
         )
 
     def _show_connection_check_message(self):
```

luzpaz, 2024-12-31:
Related ?  
* https://github.com/FreeCAD/FreeCAD-snap/issues/149

luzpaz, 2025-01-06:
The `connect failed: No such file or directory` output is missing elaboration. It needs more context