FreeCAD/FreeCAD #14310: FreeCAD 0.21.2 | Draft Import : SVG as Geometry failed

pijee, 2024-05-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When i try to import a svg file from menu (file -> import -> {choose svg file}), a dialog box appears and ask me to select module. I use __SVG as Geometry (ImportSVG)__ module and it's fail.

### Full version info

```shell
The full message error i got

12:18:42  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/freecad/Mod/Draft/importSVG.py", line 1793, in insert
    parser.parse(pythonopen(filename))
                 ^^^^^^^^^^
<class 'NameError'>: name 'pythonopen' is not defined
```
```

### Subproject(s) affected?

None

### Anything else?

I tried to modify directly /usr/lib/freecad/Mod/Draft/ImportSVG.py.

At the line 75 : 
```
# Save the native open function to avoid collisions
if open.__module__ in ['__builtin__', 'io']:
    pythonopen = open
```

I made a little ugly hack : 
```
# Save the native open function to avoid collisions
#if open.__module__ in ['__builtin__', 'io']:
pythonopen = open
```

I can now open my svg file but still have some errors : 
```
12:26:57  /usr/lib/freecad/Mod/Draft/importSVG.py:416: SyntaxWarning: invalid escape sequence '\.'
  _num = '([-+]?[0-9]*\.?[0-9]+([eE][-+]?[0-9]+)?)'
12:26:57  /usr/lib/freecad/Mod/Draft/importSVG.py:732: SyntaxWarning: invalid escape sequence '\d'
  inks_ver_pars = re.search("\d+\.\d+", inks_full_ver)
12:26:57  /usr/lib/freecad/Mod/Draft/importSVG.py:935: SyntaxWarning: invalid escape sequence '\s'
  _command = '\s*?' + _op + '\s*?' + _op2 + '\s*?'
12:26:57  /usr/lib/freecad/Mod/Draft/importSVG.py:935: SyntaxWarning: invalid escape sequence '\s'
  _command = '\s*?' + _op + '\s*?' + _op2 + '\s*?'
12:26:57  /usr/lib/freecad/Mod/Draft/importSVG.py:935: SyntaxWarning: invalid escape sequence '\s'
  _command = '\s*?' + _op + '\s*?' + _op2 + '\s*?'
12:26:57  /usr/lib/freecad/Mod/Draft/importSVG.py:938: SyntaxWarning: invalid escape sequence '\.'
  _num = '[-+]?[0-9]*\.?[0-9]+'
12:26:57  /usr/lib/freecad/Mod/Draft/importSVG.py:1620: SyntaxWarning: invalid escape sequence '\('
  _val = '\((.*?)\)'
12:26:57  /usr/lib/freecad/Mod/Draft/importSVG.py:1621: SyntaxWarning: invalid escape sequence '\s'
  _transf = _op + '\s*?' + _val
```
Despite this, i can use extrusion from part workbench and use part design with no problems.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-27:
Can you try to replicate this in the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Syres916, 2024-05-27:
This is a perfect example why we ask for full info, probability is the user has Python 3.12 installed but we don't know because it's not been given. The fix for using open as pyopen was introduced very recently in the dev build, not sure if it's been backported.

pijee, 2024-05-27:
> This is a perfect example why we ask for full info, probability is the user has Python 3.12 installed but we don't know because it's not been given. The fix for using open as pyopen was introduced very recently in the dev build, not sure if it's been backported.

Ok sorry i didn't know.

```
❯ uname -r && python --version
6.9.1-arch1-2
Python 3.12.3
```

pijee, 2024-05-27:
> Can you try to replicate this in the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

So i downloaded __[FreeCAD_weekly-builds-37464-conda-Linux-x86_64-py311.AppImage](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-37464-conda-Linux-x86_64-py311.AppImage)[FreeCAD_weekly-builds-37464-conda-Linux-x86_64-py311.AppImage](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-37464-conda-Linux-x86_64-py311.AppImage)__ and launch 2 SVG. It's works well even if the extrusion is still in shadow (i don't know if it's normal or not).

By the way, it's seems working on my machine. Great job and can't wait the new interface. Thanks !

closed 2024-05-27 by pijee