FreeCAD/FreeCAD #14312: DRAFT: Error Shape2DView

gianlucafiore, 2024-05-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
[image]
### Invoking the Shape2DView tool brings up the following error. I tried Section Planes and Levels and the alert mentioned above comes up, and it generates an empty draft object with an error character

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37464 (Git)
Build type: Release
Branch: main
Hash: db48a78d5a545bb21a5c80a9fa4b4c983fbbe951
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-27:
Are there any steps to reproduce? Does it work in the stable version?
@Roy-043 FYI

gianlucafiore, 2024-05-27:
https://drive.google.com/file/d/11YBxG2j8rRH4YZv7gNK0Y38-0bpRjdnr/view?usp=drive_link
I append test file

From the BIM workbench select a Building Level, click on the section tool, then select that created section and go to the Draft workbench, click on the Shape2DView tool and the aforementioned error arises

[image]

Roy-043, 2024-05-27:
This issue is presented in a very confusing manner. The images show a different model than the file. And the file does not produce the "This object is immutable" error, which is the main issue I think. The linked PR does fix that error however.

The images show several other errors. These should be addressed in separate issues.

gianlucafiore, 2024-05-27:
@Roy-043  Sorry for my mistake in the file link, I took another one from the folder that uploaded the tests for the issues.

Now I correct the link

PR #14319 (2024-05-28): Draft: fix _get_shapes in shape2dview.py
Fixes #14312.

The `new _get_shapes` function could return the original object shape. This caused a problem if `sh.reverse()` executed.

Note that the file in the linked issue does not produce the error as it does not contain a shape with a negative volume.
commit 524b106e (2024-05-28): Draft: fix _get_shapes in shape2dview.py (#14319)

Fixes #14312.

The `new _get_shapes` function could return the original object shape. This caused a problem if `sh.reverse()` executed.

Note that the file in the linked issue does not produce the error as it does not contain a shape with a negative volume.
closed 2024-05-28 by Roy-043