FreeCAD/FreeCAD #14314: CAM: CW/CCW/Conventional/Climb terminology consistency

jamwaffles, 2024-05-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Some CAM operations define cutting direction with CW/CCW (Clockwise/Counterclockwise) which can be confusing, along with the fact that other operations use "conventional" and "climb" verbiage, which is more commonly used in the industry and (potentially?) easier to understand the effects of. IMO it would be good to standardise on the "conventional" and "climb" direction verbiage.

A very quick audit of the UIs for all operations is as follows:

| Operation     | Terms                           |
| ------------- | ------------------------------- |
| Profile       | Direction -> CW/CCW             |
| Helix         | Direction -> CW/CCW             |
| Deburr        | Direction -> CW/CCW             |
| Pocket        | Cut Mode -> Climb/Conventional  |
| MillFace      | Cut Mode -> Climb/Conventional  |
| Pocket3D      | Cut Mode -> Climb/Conventional  |
| Threadmilling | Direction -> Climb/Conventional |
| Adaptive      | No option                       |
| Slot          | No option                       |
| Engrave       | No option                       |
| Deburr        | No option                       |
| 3D surface    | No option                       |
| Waterline     | No option                       |

I also noticed the use of "Cut Mode" and "Direction". This could be made consistent too, although perhaps to a secret third option: "Cut Direction". IMO this makes more sense when used as part of a sentence, e.g. "The first pocket operation uses the conventional cut direction".

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37464 (Git) AppImage
Build type: Release
Branch: main
Hash: db48a78d5a545bb21a5c80a9fa4b4c983fbbe951
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * freecad.gears 1.2.0
  * Assembly3 0.12.2
  * OpenTheme 2024.5.16
  * Assembly4 0.50.12
  * FeedsAndSpeeds 0.5.0
  * fasteners 0.5.21
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jamwaffles, 2024-05-28:
Thanks for double checking my (poor) work!

> however, after the initial sot is made 

I haven't been able to make a slot work properly yet, so I didn't discover this behaviour

> You listed Deburr Twice.

🤦

sliptonic, 2025-06-22:
Cthe CW/CCW/Climb/Conventional is still not consistent across all operations.  
I'm reopening the issue to track progress.

sliptonic, 2025-06-24:
We can't simply replace terminology with 'climb/conventional'  and move on.  This doesn't account for tools that don't have rotation (laser, waterjet, plasma, dragknife, pen, etc).  Users may still be opinionated about direction of travel but climb/conventional is meaningless in these cases.

I think the operation should have all the properties: direction of travel, cut mode, and cut side.  The base class should do the right thing if the user changes any of them.   Users may see properties that don't do anything but changing the ones they care about should do the right thing.

I've started writing a 'Compass' class to help with this.

```
class Compass:
    """
    A compass is a tool to help with direction so the Compass is a helper
    class to manage settings that affect tool and spindle direction.

    Settings managed:
        - Spindle Direction (from tool controller): Forward / Reverse / None
        - Cut Side: Inside / Outside
        - Cut Mode: Climb / Conventional
        - Path Direction: CW / CCW (derived)

    This class allows the user to set and get any of these properties and the rest will update accordingly.

    """

    FORWARD = "Forward"
    REVERSE = "Reverse"
    NONE = "None"
    CW = "CW"
    CCW = "CCW"
    CLIMB = "Climb"
    CONVENTIONAL = "Conventional"
    INSIDE = "Inside"
    OUTSIDE = "Outside"

    def __init__(self, tool_controller):
        self._tool_controller = tool_controller
        self._cut_side = self.OUTSIDE
        self._cut_mode = self.CLIMB
        self._path_dir = self._calculate_path_dir()

    @property
    def spindle_dir(self):
        val = getattr(self._tool_controller, "SpindleDir", self.NONE)
        if val not in (self.FORWARD, self.REVERSE, self.NONE):
            return self.NONE
        return val

    @property
    def cut_side(self):
        return self._cut_side

    @cut_side.setter
    def cut_side(self, value):
        self._cut_side = value.capitalize()
        self._path_dir = self._calculate_path_dir()

    @property
    def cut_mode(self):
        return self._cut_mode

    @cut_mode.setter
    def cut_mode(self, value):
        self._cut_mode = value.capitalize()
        self._path_dir = self._calculate_path_dir()

    @property
    def path_dir(self):
        return self._path_dir

    def _calculate_path_dir(self):
        if self.spindle_dir == self.NONE:
            return "UNKNOWN"

        spindle_rotation = self._rotation_from_spindle(self.spindle_dir)

        for candidate in (self.CW, self.CCW):
            mode = self._expected_cut_mode(self._cut_side, spindle_rotation, candidate)
            if mode == self._cut_mode:
                return candidate

        return "UNKNOWN"

    def _rotation_from_spindle(self, direction):
        return self.CW if direction == self.FORWARD else self.CCW

    def _expected_cut_mode(self, cut_side, spindle_rotation, path_dir):
        lookup = {
            (self.INSIDE, self.CW, self.CCW): self.CLIMB,
            (self.INSIDE, self.CCW, self.CW): self.CLIMB,
            (self.OUTSIDE, self.CW, self.CW): self.CLIMB,
            (self.OUTSIDE, self.CCW, self.CCW): self.CLIMB,
        }
        return lookup.get((cut_side, spindle_rotation, path_dir), self.CONVENTIONAL)

    def report(self):
        report_data = {
            "spindle_dir": self.spindle_dir,
            "cut_side": self.cut_side,
            "cut_mode": self.cut_mode,
            "path_dir": self.path_dir,
        }

        Path.Log.debug("Machining Compass config:")
        for k, v in report_data.items():
            Path.Log.debug(f"  {k:15s}: {v}")
        return report_data

```

Connor9220, 2025-06-24:
Are you saying we're going to stick with CW/CCW terminology? Or will that be something that changes based on the machine type or something?  Using our laser, I've never cared about the direction of travel, nor do I care with my Vinyl Cutter.

Connor9220, 2025-06-24:
This almost needs to be based on either the operation type OR the tool type OR machine type.  

I don't like machine type, as machines can be multi function.  They have a laser and a spindle and you can do both without changing any setups.  

Operation type might be the way to go, but some operations might be shared?

Looks like F360 uses operation to determine based on operation.. 

For Milling (2D/3D Contour, Adaptive, etc.)
Climb Milling
Conventional Milling

For Laser / Waterjet / Plasma These use the "2D Profile" operation, but the options are tailored:
"Sideways Compensation": set to Left, Right, or Center of the line.
"Reverse Cutting Direction": checkbox that simply flips CW/CCW pathing.

sliptonic, 2025-06-24:
> Are you saying we're going to stick with CW/CCW terminology? Or will that be something that changes based on the machine type or something? Using our laser, I've never cared about the direction of travel, nor do I care with my Vinyl Cutter.

I'm suggesting that after we complete the work) if you look into the property pane you would see properties for 
- cut mode (climb/conventional)
- cut side (Inside/Outside)
- direction (CW/CCW)

User could switch any of them at any time.  The others would update automatically.  
So a laser user would still see a 'cut mode' and could change it.   But probably, he's looking for direction and will change that.

The UI task panel could be more intelligent.  It could show appropriate controls based on the TC or something.

sliptonic, 2025-06-24:
Cut side should be managed independently and not change if the user switches direction or mode, of course.

> Direction is referring to the travel direction?

Yes

sliptonic, 2025-06-24:
> [@sliptonic](https://github.com/sliptonic) I see nothing wrong with this approach. Makes sense to me.

Potential downside is confusion.  A naive user explicitly sets direction of travel and then sets cut mode.  Direction of travel changes.  Switching property values on a user is generally ill-advised.  But I don't see an alternative.  We certainly don't want to pop up a notice every time.

knipknap, 2025-06-24:
One drop-down could show both.

- CW (Climb cut)
- CCW (Conventional cut)

The available selections could update based on the spindle direction.

sliptonic, 2025-06-24:
> I think showing both cut mode and direction would be confusing. I'm hoping we can do one or the other based on something as you suggest. Maybe like you said, the TC or the Operation can hide/show the appropriate one..

Help text on the properties could make them aware that they are linked.  Task panels can help a lot.  

> One drop-down could show both.
> 
>     * CW (Climb cut)
> 
>     * CCW (Conventional cut)
> 
> 
> The available selections could update based on the spindle direction.

Not just spindle direction.  Cut side too.   It gets confusing fast.  Changing the contents of a drop-down also is very unfriendly to users.

Connor9220, 2025-06-24:
> One drop-down could show both.
> 
> * CW (Climb cut)
> * CCW (Conventional cut)
> 
> The available selections could update based on the spindle direction.

That won't work.  That's the problem today.  CW doesn't always = Climb Cut.. nor does CCW = Conventional..

sliptonic, 2025-06-24:
> A separate tab might be a bit heavy though and slow down more experienced users.

We usually talk about UI efficiency in terms of 'number of mouse clicks'.  Obviously that's not the whole story otherwise we could just put every control on one screen and it would be perfectly efficient.  There's also the time it takes the user to find the right control and understand its usage.  Tabs work well for this.  They hide controls that aren't relevant to the current task.  They also help avoid mistakes by placing related controls close together.

So there's a sweet spot between adding clicks and hiding controls.  We should  try to divide things up where we think the user's intention and attention/focus is changing.

I usually start by picking operations and getting the green lines to go where I want them, First in XY.
Next I check my Z positions and step downs.
Once that is done, I check to make sure they're going in the right direction. 
Then I make sure the entry and exit moves are right and configure holding tags if needed.
I hate fiddling around with Feeds and Speeds so I configure that last.

Those big steps roughly correspond to operations, 
Depths tabs
Operation direction
Dressups
TC configuration.

Of course, No two users will do things the same way.  Still roughly chunking up the work should be possible.

Connor9220, 2025-06-24:
We may be over thinking things here.

The proper place for this IS where it's at now.. under the Operation panel on the Tasks. 
We just need to have 2 Drop Downs, only one shows at any given time.  We just need to figure out the best way to show which drop down.

Having two options in the Property view is probably fine as long as they toggle each other correctly based on inside/outside and spindle direction.

bombela, 2025-07-16:
Just have a group of two radios buttons (two boolean mutually exclusive). One "climb (chips fly behind)" one "traditional (chips fly forward)".

Connor9220, 2025-11-14:
So, the biggest issue here is, some CNC machines will be CW / CCW vs Climb / Conventional 

Drag Knife
Laser
Plasma
EDM (We don't really have support for this I don't think?)
Foam Cutter (Don't really have support for Hot wire cutter)

In short, anything that doesn't involve a rotary tool.  

The new tool library has mixings called CuttingToolMixin and RotaryToolMixin.  Anything using these tools will have a Climb / Conventional toolpath vs CW / CCW

We simply need to add the following toolbit types: Laser, Plasma, Drag Knife, and Waterjet

We don't need to do this at a machine level.  Some machines may have multiple types of tools.  I can use a Drag Knife in my router for example.

dbtayl, 2025-11-22:
I'm decidedly in the camp that at least mills should use climb/conventional as the terminology. That's the standard practice everywhere, and not doing so is asking for problems- both because users then need to think about (and maybe screw up) converting in their heads from CW/CCW+inside/outside to climb/conventional, and because it's (from a user's perspective) needlessly different from every other software.

For other tools, I don't really have any opinions- I don't use them, so my opinions there would be uninformed at best.

luzpaz, 2026-09-24:
What's left to do here ?

PR #14364 (2024-06-12): CAM: Rename CW/CCW to Climb/Conventional for consistency
As per title. I don't write much Python so I may have missed something silly, but the tests pass (locally anyway).

The translations were done with a naive find and replace. My editor stripped trailing whitespace which I can undo if it's significant, otherwise apologies for the diff noise. I'm unsure how to test if the translations are still correct.

Closes #14314