FreeCAD/FreeCAD #14321: BIM: getTreeViewItem in BimViews.py fails if the comma is the decimal separator

Roy-043, 2024-05-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The code probably also fails for imperial unit systems. The Quantity conversion seems unnecessary. Why not just use `lvH=obj.Placement.Base.z` and `lvH=obj.Elevation`?

### Full version info

```shell
V0.22 dev
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hoshengwei, 2024-05-28:
I think it is correct.
the variable `lvHStr` is trying to show the level height with User's default unit , so it use Quantity conversion and UserString to show it

but `lvH` is only the number for sorted the level height after.
so maybe use  the  `obj.Placement.Base.z`   to pass the number directly is better than the string conversion then replacement and then float and round.

yorikvanhavre, 2024-05-30:
Indeed this could be simplified.. See PR below

PR #14387 (2024-06-24): BIM: Simplificed BimViews.getTreeViewItem
Fixes #14321
commit 727ff9a0 (2024-06-24): BIM: Simplificed BimViews.getTreeViewItem - fixes #14321
closed 2024-06-24 by yorikvanhavre