FreeCAD/FreeCAD #14327: TechDraw: access violation error on dimension undo/redo

maxwxyz, 2024-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Creating a dimension with the auto dimension tool or a single dimension tool, undoing this and redoing this creates an error but the dimension is visible again.

Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/87a85568-6f65-4a04-b70c-02e6792d7e87

Output from report view: 
`<Exception> Access violation`

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.16
  * Curves 0.6.37
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14
```

### Subproject(s) affected?

Techdraw

### Anything else?

Maybe related to:
- https://github.com/FreeCAD/FreeCAD/issues/13618

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-11-15:
cc @WandererFan

kadet1090, 2025-08-13:
@theo-vt maybe interesting to you

theo-vt, 2025-08-14:
I remember touching on dimensions in #22795 I'll check tomorrow if I can reproduce the error and if that PR would fix it

theo-vt, 2025-08-14:
Ok so I cannot reproduce the exact bug on my system on 1.1dev, but the dimension does not reappear at the right place, which is fixed by https://github.com/FreeCAD/FreeCAD/pull/22795

maxwxyz, 2025-11-15:
No issue anymore, works!

closed 2025-11-15 by maxwxyz