FreeCAD/FreeCAD #14328: TechDraw: Dimension tool places dimension automatically with preselection

maxwxyz, 2024-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The smart dimension tool in techdraw places the dimension automatically when you have a preselection when you move around. The tool stays active.
Steps to reproduce:
1. select a line
2. dimension tool
3. move around the page without clicking

Demo, first without preselection (working), second attempt with a preselection (I did not click):

https://github.com/FreeCAD/FreeCAD/assets/6246609/3c182097-f4ef-4e22-b154-7f1d5e6e71f1

@PaddleStroke FYI

I have this mode enabled but it is buggy when I uncheck this too.
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.16
  * Curves 0.6.37
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-05-29:
This one is having me running in circles.

maxwxyz, 2024-05-29:
Can you reproduce this?

PaddleStroke, 2024-05-29:
Oh yes and its annoying me greatly. The command is being commited, creating this unwanted constraint, but I cannot find from where and why.

PaddleStroke, 2024-05-29:
Finally found the reason. There is something called AutoTransaction that is commiting by itself.
One has to deactivate it in the command activated function to get rid of it...
`    App::AutoTransaction::setEnable(false);`

Which sketcher smart dimension command has. So I must have had the exact same issue on sketcher and my brain just totally forgot about it.