FreeCAD/FreeCAD #14330: Inspection: Annotation not working

maxwxyz, 2024-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the Inspection WB trying to annotate an inspection creates a long, seemingly infinite line. No annotation is visible, also not in the tree view:
[image]

When the annotation is places, I cannot save the file. A click on save or save as does nothing.

File:

[inspection.zip](https://github.com/FreeCAD/FreeCAD/files/15464704/inspection.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes 1.0.9
  * Curves 0.6.36
  * fasteners 0.5.21
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-09:
@maxwxyz anyone we can ping regarding Inspection WB ?

maxwxyz, 2024-08-10:
No idea.

closed 2026-08-29 by Lgt2x