FreeCAD/FreeCAD #14331: Draft/BIM: Entered data get's overwritten when trying to click OK

maxwxyz, 2024-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This seems to be an issue with Draft and BIM commands when the overlay task panel with transparency is active.
When entering a value in the active task, I want to confirm it with the OK button. When moving the mouse up to the OK button, it overwrites the values in the input boxes as the OK button is not in the same field and the transparency enables click through (which is good for the task dialog in general and shouldn't be changed).
Maybe the buttons can either be moved down o a field with background or place a background under the buttons.

Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/5574019f-faf7-4e30-98e4-9ba13701f83b

@Roy-043 @yorikvanhavre FYI

@kadet1090 maybe you know a workaround. I guess it's the same workaround which would be needed for: https://github.com/FreeCAD/FreeCAD/issues/13350

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.16
  * Curves 0.6.37
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-05-28:
Hmm it's indeed a serious problem. There are workarounds though:

1. Pressing Enter will do the same as clicking OK
2. Set a mouse delay long enough in Draft preferences

But it's basically for that kind of reasons why there is a "Add point" button right under X,Y,Z in many other Draft tools (Line, etc...). Maybe we could add that button here too?

maxwxyz, 2024-05-28:
I'd be in favor of an add point button

kaiwas, 2024-05-28:
I will express my humble opinion.
data in input fields that depend on the position (movement) of the mouse in 3D view causes more inconvenience than benefit.
Maybe make "mouse delay" the default for all operations. And when you need to get mouse-dependent values, activate the feature with a modifier key.
For me the current situation is a real pain. It’s good that the problem is at least partially solved in the draft workbench.

yorikvanhavre, 2024-05-30:
> Maybe make "mouse delay" the default for all operations. 

It is so already.

>And when you need to get mouse-dependent values, activate the feature with a modifier key.

That would be extremely counter-productive, specially for people used to an AutoCAD-like workflow. If you want to work on such feature, no problem, but it should be optional.

kaiwas, 2024-05-30:
> That would be extremely counter-productive, specially for people used to an AutoCAD-like workflow. If you want to work on such feature, no problem, but it should be optional.

I used AutoCAD and everything works there differently than in FreeCAD. (I think I'll have to install AutoCAD again for the demonstration)
Coordinates are entered with the mouse there until we activate the input fields. Always. In all instruments.

Once data is entered into the fields, mouse movement has no effect.

Mouse delay is now implemented, but not everywhere. (only in the workbench draft for some operations).
Take even this specific example. The view shows that the data has been entered into the fields. But then, due to the fact that there is a gap between the windows, the cursor hits the 3D window and the entered data “jumps”.
That is, data in the field is lost due to short-term mouse movement.

kaiwas, 2024-05-30:
@yorikvanhavre  Example from AutoCAD.
Once data is entered, it is never reset due to mouse movement.
If I have entered data into the input fields, I can return to entering values ​​with the mouse by pressing ESC. And nothing else. And it is right.

https://github.com/FreeCAD/FreeCAD/assets/32273833/e748f07a-7666-4454-9694-2e3e03356687

kaiwas, 2024-05-30:
The wall is just one of many examples where mouse delay still doesn't work. You have to hold your breath and try very hard not to accidentally move the mouse into the 3D view. Because otherwise everything will be reset.

[image]

Freecad has many features. But it is precisely this feature that irritates me the most (after AutoCAD, ArchCAD, and all other CAD systems that I have ever tried)

maxwxyz, 2024-05-31:
@yorikvanhavre @kaiwas The delay is a bad workaround in my opinion. It is not clear in which order the user inputs data and if the user wants to change something later after entering the data.

Proposal:
1. Before entering any data manually, it shall be responsive/interactive as today.
2. When data is entered, the field should get "locked" in and should not change with mouse movement, ever.
If there are 3 fields, only the one which had a manual input should get locked. Then you can still override the other 2 fields interactively with mouse movement but one coordinate is fixed. 
This way you can make e.g. a straight wall but define the length in the 3D view interactively.
3. There should be a button below the input fields, to pick a point. If pressed, the user can pick a point in 3D view, all 3 fields are not locked if the button is pressed, so change on mouse movement is active. After left clicking, the point is locked in as today.

@FreeCAD/design-working-group FYI if someone wants to weigh in.

kaiwas, 2024-05-31:
> The delay is a bad workaround in my opinion. It is not clear in which order the user inputs data and if the user wants to change something later after entering the data.

Yes, you very accurately described the principle that is used in all CAD systems. To disable "blocking" of a field, ESC is most often used.
There is no time delay. I talked about this before, but Yorik decided to go this route. And in general this can work if  add an ESC key to cancel.

In addition, data in fields is lost not only in ARCH/Draft. Similar behavior also occurs in sketcher.

Essentially, two “layers of behavior” are needed.
The first is the mouse, the second is the input field. The field must be given priority if enabled. Exiting “field mode” is most often canceled by the ESC.

yorikvanhavre, 2024-05-31:
FWIW I have no problem introducing another behaviour, as long as it can be disabled and go back to the current behaviour

kaiwas, 2024-05-31:
> FWIW I have no problem introducing another behaviour, as long as it can be disabled and go back to the current behaviour. But again, I think what you both describe is what happens when the mouse delay is set to something "permanent" (ex: 1hour). Shoudn't we then make THAT the default?

I don't think there is any time limit in other programs. But so be it. (the time approach itself does not interfere in any way if you set a high value).
For now, the main question is that the mouse delay only works in some of the tools, but it is needed absolutely everywhere where there are input fields and input of coordinates with the mouse.

yorikvanhavre, 2024-05-31:
> I don't think there is any time limit in other programs

(Which is not a reason why we can't ;) ) but in any case, if the time limit is very high it's not a time limit anymore. Consider it an easy way to have all kinds of behaviours possible with only one setting.

Things are very differently implemented in Draft than in other parts of FreeCAD, and I don't think there is an easy way to implement this once and it would work FreeCAD-wide, as each workbench implements their own UI separately.

What we could do, though is start with Draft:

1) Identify Draft UI elements that need this
2) Implement this everywhere in Draft
3) Move the preferences setting somewhere outside Draft
4) In the future, start porting other UIs in other WBs to use it

kaiwas, 2024-05-31:
> What we could do, though is start with Draft:
I suggest moving back to this topic (I have it in my browser bookmarks on all computers).
https://github.com/FreeCAD/FreeCAD/pull/12624
Or open a discussion.
I will analyze all the tools and let you know how they work.

> (Which is not a reason why we can't ;) )

Agree. I see no reason why this cannot be done.

> Things are very differently implemented in Draft than in other parts of FreeCAD, and I don't think there is an easy way to implement this once and it would work FreeCAD-wide, as each workbench implements their own UI separately.

This is sad, but probably someday it will be necessary to solve this problem globally. In my personal opinion, this problem greatly discourages everyone who is planning to use FreeCad.
I've been experimenting with the program for almost a year now and still can't get used to data loss. For me this is the main problem of the program.

obelisk79, 2024-05-31:
I would propose the behavior work similar to what Max proposed. 

When one of those input fields has active 'focus' for keyboard input, mouse position data is ignored. 
Reenable mouse position data as field inputs by clicking out of the field (aka input focus is lost).

Making it behave like this should be relatively simple, and predictable and not result in any loss of existing capability, yes/no?

maxwxyz, 2024-05-31:
I guess this solution wouldn't solve the issue when the mouse position data is enabled when leaving the fields, as I still need to get to the OK button or click somewhere else in the dialog or in other panels (see the video in the OP).
IMO mouse input should be ignored for the field if there was a manual keyboard input, as long as the user would click on a following button: pick point.

obelisk79, 2024-05-31:
That is what I mean by input focus. You are thinking about mouse focus. If the blinking cursor is active in a field from typing with the keyboard (ie input focus), then mouse input for those fields is disabled. As soon as the fields no longer are capturing keyboard input, mouse input resumes.

kadet1090, 2024-05-31:
Relying on focus won't solve anything as you can have only one input focused at once and here you have 3 distinct inputs that you don't want touch if the value was entered manually. I totally agree with max here - once there was a keyboard input to the field it should not be changed by mouse until user manually allows it by button or for example clearing the field.

obelisk79, 2024-05-31:
I better understand now, and agree with Kadet and Max's suggestions for behavior.

kaiwas, 2024-05-31:
> IMO mouse input should be ignored for the field if there was a manual keyboard input, as long as the user would click on a following button: pick point.

I agree, but there are specific tools.
For example, a line in a sketcher.
There are two main fields (length and angle).
The expected behavior after entering the length is that by switching the angle, the length will be fixed, but the angle can still be changed (until I entered data in the angle field).
A similar example is in my video from AutoCAD.
In general, the task is complex and interesting. I understand that this cannot be solved in simple ways. But I really hope this happens. Because the current situation is very sad.

closed 2026-06-13 by Roy-043