FreeCAD/FreeCAD #14332: Measure does not handle selecting link elements

NomAnor, 2024-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Activating Std_Measure and then selecting elements (eg. vertices) of Link objects will measure the elements of the original, not of the link.
[image]
I have clicked on the red vertices but the properties are linked to the original vertices.

[
[image]
](url)
If I manually change the properties, with the "Sync sub-object selection" option checked in the selection dialog, the correct measurement is shown.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37330 (Git)
Build type: Debug
Branch: main
Hash: a094bf0df7e2b3e5c5b2e3b158a9040a6474892f
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.3.0
Locale: German/Germany (de_DE)
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.62.0
  * Assembly3.backup1715897154.0970206 0.11.3 (Disabled)
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.2.0
  * fasteners 0.4.21
```

### Subproject(s) affected?

Core

### Anything else?

This looks like a bug in the selection handling code, not the calculation code.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-05-28:
@hlorus FYI

NomAnor, 2024-09-02:
This but is still there in a fresh master build:
```
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38513 (Git)
Build type: Release
Branch: main
Hash: 7c797b5a10c19ae55e317c4ff9a02823a4b4dc0b
Python 3.12.5, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.1, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.62.0
  * Assembly3.backup1715897154.0970206 0.11.3 (Disabled)
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.2.0
  * fasteners 0.5.27

```

[image]

maxwxyz, 2024-09-02:
@hlorus FYI, seems still to be an issue.

furgo16, 2025-06-30:
This is still relevant

closed 2025-08-31 by maxwxyz

maxwxyz, 2025-10-31:
so is the issue fixed?

commit 433dcec0 (2024-06-07): MeasureGui: Don't resolve selection objects in TaskMeasure

Fixes #14332