FreeCAD/FreeCAD #14333: Arc length dimension doesn't snap to other dimensions

prokoudine, 2024-05-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Select the new auto dimension tool in TD
2. Click on an arc and drag out a dimension
3. Press M to toggle the mode until you get arc length
4. Try to snap it to other existing dimensions
5. No snapping occurs

https://github.com/FreeCAD/FreeCAD/assets/57467/748b4130-9546-490c-821c-8f644f3a593c

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.37455 (Git) AppImage
Build type: Release
Branch: flavor
Hash: 01e9de13fa3b534aba6588f1a7e6fa9748249899
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Ondsel-Lens.backup1714137477.6968067 2024.4.16.01 (Disabled)
  * OpenDark.backup1714484818.1582093 2024.3.13 (Disabled)
  * Ondsel-Lens.backup1714484818.5582054 2024.4.16.01 (Disabled)
  * OpenTheme 2024.4.20
  * Lattice2 1.0.0
  * Ondsel-Lens.backup1714855588.2689874 2024.4.16.01 (Disabled)
  * Curves 0.6.37
  * freecad.gears 1.2.0
  * OpenDark.backup1714137477.059234 2024.3.13 (Disabled)
  * Ondsel-Lens 2024.5.6.01
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-05-28:
This is very similar to one of the issue raised in https://github.com/FreeCAD/FreeCAD/issues/14325

The issue is that currently dimensions of different types do not snap together. That's why your arc angle is not snapping.

We could try to improve by checking if the dimension is parallel rather than of the same type. I'm currently on it.

maxwxyz, 2024-05-28:
Also related: 
- https://github.com/FreeCAD/FreeCAD/issues/13831

closed 2024-06-03 by WandererFan